kenzok8/openwrt-packages #3: 无法应用主题

caoli5288, 2020-04-04:
原版openwrt x86-64 snapshot。

luci-theme-opentomato

```
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'view'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/opentomato/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/opentomato/he..."]:20: attempt to call field 'node_childs' (a nil value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/opentomato/he..."]:20: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:694: in function 'include'
	[string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:694: in function 'include'
	[string "/usr/lib/lua/luci/view/view.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:857: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:441: in function </usr/lib/lua/luci/dispatcher.lua:440>
```

luci-theme-opentomcat

```
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'view'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/opentomcat/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/opentomcat/he..."]:23: attempt to call field 'node_childs' (a nil value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/opentomcat/he..."]:23: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:694: in function 'include'
	[string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:694: in function 'include'
	[string "/usr/lib/lua/luci/view/view.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:857: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:441: in function </usr/lib/lua/luci/dispatcher.lua:440>
```

ATMaterial 不会报错，但是应用无效果。

ClayMoreBoy, 2020-04-10:
snapshot可用?

kenzok8, 2020-04-13:
18.06 luci适配

closed 2023-01-30 by kenzok8