kenzok8/openwrt-packages #11: ssr+无法运行

iwowcn, 2020-04-28:
ssh手动启动提示：
root@OpenWrt:~# /etc/init.d/shadowsocksr start
-ash: /etc/init.d/shadowsocksr: Permission denied
控制台启动显示“ShadowsocksR Plus+ 未运行”，停用再启用也无效。

kenzok8, 2020-04-28:
我自己编译没问题，Lean与Lienol源码都没问题

iwowcn, 2020-04-28:
那就奇怪了，我用lean的源码，昨天还是好的，刚刚编译了一次，去init.d下面查看，ssr和阿里ddns都没有执行权限，手动加上之后各种报错。

iwowcn, 2020-04-28:
> […](#)

那就奇怪了，我用lean的源码，昨天还是好的，刚刚编译了一次，去init.d下面查看，ssr和阿里ddns都没有执行权限，手动加上之后各种报错。

kenzok8, 2020-04-28:
chmod +x /etc/init.d/shadowsocksr
chmod +x /usr/bin/ssr-rules
chmod +x /usr/bin/ss-redir
试试

iwowcn, 2020-04-28:
我刷回26晚上编译的固件了，ssr+正常，去看了一下/init.d目录，ssr权限正常，但是aliddns没有可执行权限，尝试手动添加，命令和web都不能启动阿里ddns。

iwowcn, 2020-04-28:
> chmod +x /etc/init.d/shadowsocksr
> chmod +x /usr/bin/ssr-rules
> chmod +x /usr/bin/ss-redir
> 试试

我刷回26晚上编译的固件了，ssr+正常，去看了一下/init.d目录，ssr权限正常，但是aliddns没有可执行权限，尝试手动添加，命令和web都不能启动阿里ddns。

kenzok8, 2020-04-28:
没这个反馈

shangyanjin, 2020-04-30:
1.控制台启动显示“ShadowsocksR Plus+ 未运行”，停用再启用也无效。
2.ssh手动启动提示：
root@OpenWrt:~# /etc/init.d/shadowsocksr start
-ash: /etc/init.d/shadowsocksr: Permission denied
3.shell解决
chmod +x /etc/init.d/shadowsocksr
chmod +x /usr/bin/ssr-rules
chmod +x /usr/bin/ss-redir
4.有没有办法?

kali98, 2020-05-02:
我也遇到了，我是加入feed文件里的，编译出来也提示上面一样的，有解决办法么

shangyanjin, 2020-05-05:
有没有打包好的 ？给个链接。

Henrysilently, 2020-05-07:
> 我也遇到了，我是加入feed文件里的，编译出来也提示上面一样的，有解决办法么

chmod +x /etc/init.d/shadowsocksr
chmod +x /usr/bin/ssr-rules
chmod +x /usr/bin/ss-redir
使用以上指令就可以使用了

closed 2023-01-30 by kenzok8