kenzok8/openwrt-packages #14: 请问，假如我只想使用库里面的一个插件语法怎么写

authon, 2020-05-13:
git clone https://github.com/kenzok8/openwrt-packages/luci-app-passwall package/luci-app-passwall
这样可以吗

kenzok8, 2020-05-13:
svn co https://github.com/kenzok8/openwrt-packages/trunk/luci-app-passwall

authon, 2020-05-13:
好的，谢谢，刚刚摸索了一下svn命令

authon, 2020-05-15:
> […](#)

大佬，知道如何同步更新他人的库到自己库里面的某个文件夹吗

liuran001, 2020-09-06:
> > […](#)
> 
> 大佬，知道如何同步更新他人的库到自己库里面的某个文件夹吗

子模块可以做到，但是不建议，因为直接clone库子模块会是空文件夹
建议用GitHub Action自动同步，可以参考[我的项目](https://github.com/liuran001/openwrt-packages)

authon, 2020-09-07:
> > > […](#)
> > 
> > 
> > 大佬，知道如何同步更新他人的库到自己库里面的某个文件夹吗
> 
> 子模块可以做到，但是不建议，因为直接clone库子模块会是空文件夹
> 建议用GitHub Action自动同步，可以参考[我的项目](https://github.com/liuran001/openwrt-packages)

请问你的action代码是放在哪里的，在你项目里面没有找到workfollows目录