kenzok8/openwrt-packages #21: 建议passwall添加老版本选项

WROIATE, 2020-06-04:
由于passwall无法支持ipv6纯地址形式，老版passwall使用域名有一个选项是“使用ipv6”，可以让域名解析为ipv6进行连接
最新版passwall取消了“使用ipv6”这一选项，导致无法使用纯ipv6主机
建议分一个old文件夹专门放老版供需要的人选

kenzok8, 2020-06-04:
git clone https://github.com/kenzok8/passwall.git

WROIATE, 2020-06-04:
> […](#)

感谢！

WROIATE, 2020-06-04:
@kenzok8 使用上面这个源会出现编译错误，请问是为什么？我已经把3.6-41文件给换成这个了
```
make[3]: Entering directory '/home/j_wroiate/op19/feeds/kenzo/luci-app-passwall'
rm -rf /home/j_wroiate/op19/build_dir/target-x86_64_musl/luci-app-passwall-3.6/.pkgdir/luci-app-passwall.installed /home/j_wroiate/op19/build_dir/target-x86_64_musl/luci-app-passwall-3.6/.pkgdir/luci-app-passwall
mkdir -p /home/j_wroiate/op19/build_dir/target-x86_64_musl/luci-app-passwall-3.6/.pkgdir/luci-app-passwall
install -d -m0755 /home/j_wroiate/op19/build_dir/target-x86_64_musl/luci-app-passwall-3.6/.pkgdir/luci-app-passwall/etc/config
install -m0600 ./root/etc/config/passwall /home/j_wroiate/op19/build_dir/target-x86_64_musl/luci-app-passwall-3.6/.pkgdir/luci-app-passwall/etc/config/passwall
install -m0600 ./root/etc/config/passwall_show /home/j_wroiate/op19/build_dir/target-x86_64_musl/luci-app-passwall-3.6/.pkgdir/luci-app-passwall/etc/config/passwall_show
install: cannot stat './root/etc/config/passwall_show': No such file or directory
Makefile:152: recipe for target '/home/j_wroiate/op19/build_dir/target-x86_64_musl/luci-app-passwall-3.6/.pkgdir/luci-app-passwall.installed' failed
make[3]: *** [/home/j_wroiate/op19/build_dir/target-x86_64_musl/luci-app-passwall-3.6/.pkgdir/luci-app-passwall.installed] Error 1
make[3]: Leaving directory '/home/j_wroiate/op19/feeds/kenzo/luci-app-passwall'
time: package/feeds/kenzo/luci-app-passwall/compile#0.12#0.04#0.17
package/Makefile:111: recipe for target 'package/feeds/kenzo/luci-app-passwall/compile' failed
make[2]: *** [package/feeds/kenzo/luci-app-passwall/compile] Error 2
make[2]: Leaving directory '/home/j_wroiate/op19'
package/Makefile:107: recipe for target '/home/j_wroiate/op19/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/j_wroiate/op19/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/j_wroiate/op19'
/home/j_wroiate/op19/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

sxx1314, 2020-06-05:
从 log里看 40到41 仅仅是添加了一个 acl.d  所以应该不存在ipv6的修改  应该是更早期的版本  同理我用我以前的40固件也确实没有发现这个选项

> […](#)

从 log里看 40到41 仅仅是添加了一个 acl.d  所以应该不存在ipv6的修改  应该是更早期的版本  同理我用我以前的40固件也确实没有发现这个选项

kenzok8, 2020-06-05:
登陆软路由--网络选项--DHCP/DNS--高级设置里 ，把“禁止解析 IPv6 DNS 记录”勾选去掉试试

kenzok8, 2020-06-05:
你是git clone下载的？如果是copy的记得给passwall 权限 775就行了

WROIATE, 2020-06-07:
> 你是git clone下载的？如果是copy的记得给passwall 权限 775就行了

不是这个原因，是因为这份文件里少了几个文件，包括passwall_show，这个文件在3.6-41里有在你给的3.6-40里没有

closed 2020-07-06 by WROIATE