kenzok8/openwrt-packages #27: gost报错

zerofo, 2020-06-14:
最近一周action build5 6次了。真的不行
`Collected errors:
Configuring luci-app-gost.
 * check_data_file_clashes: Package libustream-openssl20150806 wants to install file /home/runner/work/lede/lede/build_dir/target-x86_64_musl/root-x86/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls20150806
 * opkg_install_cmd: Cannot install package libustream-openssl20150806.
 * check_data_file_clashes: Package busybox wants to install file /home/runner/work/lede/lede/build_dir/target-x86_64_musl/root-x86/sbin/insmod
	But that file is already provided by package  * ubox
 * check_data_file_clashes: Package busybox wants to install file /home/runner/work/lede/lede/build_dir/target-x86_64_musl/root-x86/sbin/lsmod
	But that file is already provided by package  * ubox
 * check_data_file_clashes: Package busybox wants to install file /home/runner/work/lede/lede/build_dir/target-x86_64_musl/root-x86/sbin/modprobe
	But that file is already provided by package  * ubox
 * check_data_file_clashes: Package busybox wants to install file /home/runner/work/lede/lede/build_dir/target-x86_64_musl/root-x86/sbin/rmmod
	But that file is already provided by package  * ubox
 * opkg_install_cmd: Cannot install package busybox.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-adguardhome:
 * 	AdGuardHome
 * opkg_install_cmd: Cannot install package luci-app-adguardhome.
 * check_data_file_clashes: Package luci-theme-argon_new wants to install file /home/runner/work/lede/lede/build_dir/target-x86_64_musl/root-x86/etc/uci-defaults/30_luci-theme-argon
 Prepare artifact`

kenzok8, 2020-06-14:
少了gost一个文件，已经补上了，抱歉

zerofo, 2020-06-15:
> 少了gost一个文件，已经补上了，抱歉
现在去试试
没啥抱歉 感谢你的付出!. 我只是一个伸手党.   只是build了很几次都不行,希望得到解决, 语出简单, 如有冒犯请多包含.

taowazi007, 2020-07-01:
/etc/init.d/gost加个775的权限

taowazi007, 2020-07-09:
gost 不能自动启动，需要手动保存&应用一次才能启动