kenzok8/openwrt-packages #31: 原版openwrt能用吗？

RayTsanger, 2020-06-23:
大神好，官方原版openwrt能用吗？先下载以来，再下载这个，再编译可以吗？

kenzok8, 2020-06-23:
原理是可以的，但是要很多设置，一时也不能细述，不能保证绝对能过，一般的办法是用Lean源码下package里的lean文件复制进openwrt/package下、然后加上我的插件包，依赖包，这样编译就行了，但是你也可以挑战下，自己去试试

RayTsanger, 2020-06-24:
试了下，都没成功啊，主要我的erx刷两位大神源码编译的固件，没有原版系统快，看来还是出手再换回软路由吧，谢谢您！

smalldickman, 2020-08-14:
> 原版的可以用啊，我的新路由3就用的这个和small，编译没问题。有点小报错需要折腾一下

可以成功编译，就是luci界面显示没有权限认证之类的不能正常显示，比如官方的原版docker和luci界面就不能用（编译两次一次加额外包一次官方原味包都不行）。。。。说是要增加那个json文件，但是不会编写就没法了（模仿添加了几个json文件可以正常显示，例如网易解锁，luci界面重启下就又不能用了，，各种问题）

hathlife, 2020-10-13:
> 原版的可以用啊，我的新路由3就用的这个和small，编译没问题。有点小报错需要折腾一下

可以具体说一下怎么解决依赖问题的吗？谢谢

smalldickman, 2020-10-13:
> > 原版的可以用啊，我的新路由3就用的这个和small，编译没问题。有点小报错需要折腾一下
> 
> 可以具体说一下怎么解决依赖问题的吗？谢谢

例如这个仓库就是依赖：https://github.com/kenzok8/small
这个仓库是应用：https://github.com/kenzok8/openwrt-packages

在官方包目录下feeds.conf.default这个文件加入
src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git small https://github.com/kenzok8/small
这两条就满足了应用和依赖
执行   ./scripts/feeds update -a 就能把这两个仓库加载进去，
如果是其他的包，执行完  ./scripts/feeds install -a  会提示缺哪些依赖，然后找齐放在一个文件，上传到自己仓库，在feeds.conf.default这个文件夹再加一句  src-git 123 https://你的仓库地址，再执行 ./scripts/feeds update -a，你找的依赖就加进去了。。 再 ./scripts/feeds install -a 一遍  就不会提示缺依赖了

不过官方包加额外应用得一个个添加测试，加太多了，编译出错不知道那个出问题。。

hathlife, 2020-10-14:
> 例如这个仓库就是依赖：https://github.com/kenzok8/small
> 这个仓库是应用：https://github.com/kenzok8/openwrt-packages
>
>
> 在官方包目录下feeds.conf.default这个文件加入
> src-git kenzo https://github.com/kenzok8/openwrt-packages
> src-git [...]

```
make[3]: *** [Makefile:78: /home/toonaive/openwrt/build_dir/target-mipsel_24kc_musl/v2ray-plugin-1.5.0/.built
] Error 127
make[2]: *** [package/Makefile:113: package/feeds/small/v2ray-plugin/compile] Error 2
make[3]: *** [Makefile:313: /home/toonaive/openwrt/build_dir/target-mipsel_24kc_musl/v2ray-core-4.30.0/.built
] Error 127
make[2]: *** [package/Makefile:113: package/feeds/small/v2ray/compile] Error 2
make[1]: *** [package/Makefile:107: /home/toonaive/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package
_compile] Error 2
```
这个报错见过吗？

moshiyao, 2021-04-11:
> > > > 原版的可以用啊，我的新路由3就用的这个和small，编译没问题。有点小报错需要折腾一下
> > > 
> > > 
> > > 可以具体说一下怎么解决依赖问题的吗？谢谢
> > 
> > 
> > 例如这个仓库就是依赖：https://github.com/kenzok8/small
> > 这个仓库是应用：https://github.com/kenzok8/openwrt-packages
> > 在官方包目录下feeds.conf.default这个文件加入
> > src-git kenzo https://github.com/kenzok8/openwrt-packages
> > src-git small https://github.com/kenzok8/small
> > 这两条就满足了应用和依赖
> > 执行   ./scripts/feeds update -a 就能把这两个仓库加载进去，
> > 如果是其他的包，执行完  ./scripts/feeds install -a  会提示缺哪些依赖，然后找齐放在一个文件，上传到自己仓库，在feeds.conf.default这个文件夹再加一句  src-git 123 [https://你的仓库地址，再执行](https://%E4%BD%A0%E7%9A%84%E4%BB%93%E5%BA%93%E5%9C%B0%E5%9D%80%EF%BC%8C%E5%86%8D%E6%89%A7%E8%A1%8C) ./scripts/feeds update -a，你找的依赖就加进去了。。 再 ./scripts/feeds install -a 一遍  就不会提示缺依赖了
> > 不过官方包加额外应用得一个个添加测试，加太多了，编译出错不知道那个出问题。。
> 
> ```
> make[3]: *** [Makefile:78: /home/toonaive/openwrt/build_dir/target-mipsel_24kc_musl/v2ray-plugin-1.5.0/.built
> ] Error 127
> make[2]: *** [package/Makefile:113: package/feeds/small/v2ray-plugin/compile] Error 2
> make[3]: *** [Makefile:313: /home/toonaive/openwrt/build_dir/target-mipsel_24kc_musl/v2ray-core-4.30.0/.built
> ] Error 127
> make[2]: *** [package/Makefile:113: package/feeds/small/v2ray/compile] Error 2
> make[1]: *** [package/Makefile:107: /home/toonaive/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package
> _compile] Error 2
> ```
> 
> ~这个报错见过吗~已解决，tools文件夹缺少upx和ucl，从lean库复制一份，然后照以下修改tools/Makefile即可，注意位置要对
> 
> ```
> tools-y += ucl upx 
> tools-y += autoconf autoconf-archive automake bc bison cmake dosfstools
> tools-y += e2fsprogs fakeroot findutils firmware-utils flex gengetopt
> tools-y += libressl libtool lzma m4 make-ext4fs missing-macros mkimage
> tools-y += mklibs mm-macros mtd-utils mtools padjffs2 patch-image
> tools-y += patchelf pkgconf quilt squashfskit4 sstrip zip zlib zstd
> tools-$(BUILD_B43_TOOLS) += b43-tools
> tools-$(BUILD_ISL) += isl
> tools-$(BUILD_TOOLCHAIN) += expat gmp libelf mpc mpfr
> tools-$(CONFIG_TARGET_apm821xx)$(CONFIG_TARGET_gemini) += genext2fs
> tools-$(CONFIG_TARGET_ath79) += lzma-old squashfs
> tools-$(CONFIG_TARGET_mxs) += elftosb sdimage
> tools-$(CONFIG_TARGET_tegra) += cbootimage cbootimage-configs
> tools-$(CONFIG_TARGET_x86) += qemu
> tools-$(CONFIG_USES_MINOR) += kernel2minor
> tools-$(CONFIG_USE_SPARSE) += sparse
> 
> # builddir dependencies
> $(curdir)/upx/compile := $(curdir)/ucl/compile
> $(curdir)/autoconf/compile := $(curdir)/m4/compile
> $(curdir)/automake/compile := $(curdir)/m4/compile $(curdir)/autoconf/compile $(curdir)/pkgconf/compile $(curdir)/xz/compile
> $(curdir)/b43-tools/compile := $(curdir)/bison/compile
> ```

lean库在哪？

moshiyao, 2021-04-11:
出了问题，查不到解决办法。
make[3]: *** [Makefile:75: /***/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.0/.built] Error 1
make[3]: Leaving directory '/***/openwrt/feeds/small/v2ray-plugin'
time: package/feeds/small/v2ray-plugin/compile#1.55#1.99#177.90
make[2]: *** [package/Makefile:113: package/feeds/small/v2ray-plugin/compile] Error 2
make[2]: Leaving directory '/***/openwrt'
make[1]: *** [package/Makefile:107: /***/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/***/openwrt'
make: *** [/***/openwrt/include/toplevel.mk:227: world] Error 2