kenzok8/openwrt-packages #33: 提示缺少依赖

hanlihanshaobo, 2020-06-27:
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssrserver-python/Makefile' has a dependency on 'python', which does not exist
WARNING: Makefile 'package/lean/qt5/Makefile' has a dependency on 'libdouble-conversion', which does not exist

用lean的没警告，原生的有提示

kenzok8, 2020-06-27:
先要知道怎么编译原生固件，或者官方lean移植固件的区别！

hanlihanshaobo, 2020-06-28:
打扰了，我是小白，主要是请教下这三个依赖怎么解决，想把lean里面的直接复制过来，但是里面搜不到

kenzok8, 2020-06-28:
lede/package/lean 把整个文件夹复制进去openwrt/package/下面

hanlihanshaobo, 2020-06-28:
我就是这样操作的，还是提示，本地和actions编译都有同样的问题

kenzok8, 2020-06-28:
cp -r ~/lede/package/lean ~/openwrt/package/lean
cp -r ~/lede/package/libs/librpc ~/openwrt/package/libs
试试

hanlihanshaobo, 2020-06-29:
> cp -r ~/lede/package/lean ~/openwrt/package/lean
> cp -r ~/lede/package/libs/librpc ~/openwrt/package

> cp -r ~/lede/package/lean ~/openwrt/package/lean
> cp -r ~/lede/package/libs/librpc ~/openwrt/package/libs
> 试试

还是不行，报相同的警告

houpi, 2020-07-03:
我也有两个，一个是是 kmod-fs-exfat ，一个是 wol
也不知道是啥原因

czpwpq, 2020-09-02:
WARNING: Makefile 'package/feeds/laohu/luci-app-ssr-python-pro-server/Makefile' has a dependency on 'python', which does not exist
我也碰到了这类问题，就其中的一个包编译不过去，去掉这个包其他都能成功，但是挺实用的，就想试着编译进去，就是不成功

closed 2023-02-01 by kenzok8