kenzok8/openwrt-packages #54: SSR缺少依赖

brianzc, 2020-09-06:
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'naiveproxy', which does not exist

o-water-o, 2020-09-14:
我也是这个情况编译不通过

Eveaz, 2020-09-15:
这是由于kenzok8大佬，没有把helloworld的文件都拷过来。
先编辑feeds.conf.default
```
#src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git helloworld https://github.com/fw876/helloworld
```
然后 ./scripts/feeds update -a 以及 ./scripts/feeds install -a
再编辑feeds.conf.default
```
src-git kenzo https://github.com/kenzok8/openwrt-packages
#src-git helloworld https://github.com/fw876/helloworld
```
然后再编译。

o-water-o, 2020-09-18:
> 这是由于kenzok8大佬，没有把helloworld的文件都拷过来。
> 先编辑feeds.conf.default
>
> ```
> #src-git kenzo https://github.com/kenzok8/openwrt-packages
> src-git helloworld https://github.com/fw876/helloworld
> ```
>
> 然后 [...]

把helloworld 源直接放开，一起编译不行么

closed 2023-01-30 by kenzok8