kenzok8/openwrt-packages #55: 请教下，按照操作提示已经将 依赖包及openwrt-packages放置feeds.conf.default，但编译还是报错

kankankankankankan, 2020-09-07:
已经按照内文说的操作了，使用Actions编译，但频繁报错，实在没办法请教下
```
sed -i '$a\src-git kenzo https://github.com/kenzok8/openwrt-packages' feeds.conf.default
sed -i '$a\src-git small https://github.com/kenzok8/small' feeds.conf.default
```
也加入了配置
```
ONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ipt2socks=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Shadowsocks=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ShadowsocksR=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_V2ray=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Brook=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_kcptun=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_haproxy=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ChinaDNS_NG=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_pdnsd=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_dns2socks=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_v2ray-plugin=y		
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_simple-obfs=y
```
#### 报错日志类型1
```
2020-09-06T07:20:08.2592954Z then cat ".deps/elf32_checksum.Tpo" >> ".deps/elf32_checksum.Po"; \
2020-09-06T07:20:08.2593244Z      rm -f ".deps/elf32_checksum.Tpo"; \
2020-09-06T07:20:08.2593521Z else rm -f ".deps/elf32_checksum.Tpo"; exit 1; \
2020-09-06T07:20:08.2593656Z fi
2020-09-06T07:20:08.3076264Z if aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.3.0_musl/usr/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.3.0_musl/include/fortify -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.3.0_musl/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144 -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/elfutils-0.177=elfutils-0.177 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -DBAD_FTS=1 -c -o elf64_checksum.os -fPIC -DPIC -DSHARED -DSYMBOL_VERSIONING -MT elf64_checksum.os -MD -MP \
2020-09-06T07:20:08.3077850Z   -MF ".deps/elf64_checksum.Tpo" `test -f 'elf64_checksum.c' || echo './'`elf64_checksum.c; \
2020-09-06T07:20:08.3078058Z then cat ".deps/elf64_checksum.Tpo" >> ".deps/elf64_checksum.Po"; \
2020-09-06T07:20:08.3078388Z      rm -f ".deps/elf64_checksum.Tpo"; \
2020-09-06T07:20:08.3078712Z else rm -f ".deps/elf64_checksum.Tpo"; exit 1; \
2020-09-06T07:20:08.3078863Z fi
2020-09-06T07:20:08.3572333Z if aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.3.0_musl/usr/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.3.0_musl/include/fortify -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.3.0_musl/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144 -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/elfutils-0.177=elfutils-0.177 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -DBAD_FTS=1 -c -o libelf_crc32.os -fPIC -DPIC -DSHARED -DSYMBOL_VERSIONING -MT libelf_crc32.os -MD -MP \
2020-09-06T07:20:08.3573794Z   -MF ".deps/libelf_crc32.Tpo" `test -f 'libelf_crc32.c' || echo './'`libelf_crc32.c; \
2020-09-06T07:20:08.3574102Z then cat ".deps/libelf_crc32.Tpo" >> ".deps/libelf_crc32.Po"; \
2020-09-06T07:20:08.3574538Z      rm -f ".deps/libelf_crc32.Tpo"; \
2020-09-06T07:20:08.3574943Z else rm -f ".deps/libelf_crc32.Tpo"; exit 1; \
2020-09-06T07:20:08.3575083Z fi
2020-09-06T07:20:08.3860569Z if aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.3.0_musl/usr/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.3.0_musl/include/fortify -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.3.0_musl/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144 -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/elfutils-0.177=elfutils-0.177 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -DBAD_FTS=1 -c -o libelf_next_prime.os -fPIC -DPIC -DSHARED -DSYMBOL_VERSIONING -MT libelf_next_prime.os -MD -MP \
2020-09-06T07:20:08.3861644Z   -MF ".deps/libelf_next_prime.Tpo" `test -f 'libelf_next_prime.c' || echo './'`libelf_next_prime.c; \
2020-09-06T07:20:08.3861829Z then cat ".deps/libelf_next_prime.Tpo" >> ".deps/libelf_next_prime.Po"; \
2020-09-06T07:20:08.3862120Z      rm -f ".deps/libelf_next_prime.Tpo"; \
2020-09-06T07:20:08.3862403Z else rm -f ".deps/libelf_next_prime.Tpo"; exit 1; \
2020-09-06T07:20:08.3862536Z fi
2020-09-06T07:20:08.4111854Z if aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/runner/work/NEO2/NEO2/friendlywrt-h5/fri [... cut ...]
2020-09-06T07:20:09.1817758Z make[1]: Leaving directory '/home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt'
2020-09-06T07:20:09.1818431Z /home/runner/work/NEO2/NEO2/friendlywrt-h5/friendlywrt/include/toplevel.mk:225: recipe for target 'world' failed
2020-09-06T07:20:09.1821943Z make: *** [world] Error 2
2020-09-06T07:20:09.1832881Z Command exited with non-zero status 1
2020-09-06T07:20:09.1833118Z you take 2:58:52 to build friendlywrt
2020-09-06T07:20:09.1965846Z ##[error]Process completed with exit code 1.
```

再次感谢

kenzok8, 2020-09-08:
参考https://github.com/kenzok8/LEDE-x86_64

richaradlee, 2020-09-09:
临时解决办法，已试可以完成云编译。

Add  "src-git lienol https://github.com/Lienol/openwrt-package"  to  feeds.conf.default.

kankankankankankan, 2020-09-10:
> 参考https://github.com/kenzok8/LEDE-x86_64

感谢，这是适用于ARM的处理器么，我看是x86 64的

kankankankankankan, 2020-09-10:
> 临时解决办法，已试可以完成云编译。
> 
> Add "src-git lienol https://github.com/Lienol/openwrt-package" to feeds.conf.default.

感谢我重新试试

closed 2020-09-13 by kankankankankankan

kenzok8, 2020-09-13:
添加small得依赖包，见说明！

kenzo wu