kenzok8/openwrt-packages #56: Passwall 3.9-56 的small源缺少依赖导致编译失败

zhucaidan, 2020-09-07:
源更新的时候提示：
Create index file './feeds/small.index' 
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'ssocks', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'ssocksd', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'naiveproxy', which does not exist

编译输出时候会失败：
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-passwall:
 * 	ssocks
 * 	ssocksd
 * opkg_install_cmd: Cannot install package luci-app-passwall.
make[2]: *** [package/Makefile:68: package/install] Error 255

mkmichael, 2020-09-08:
求助遇到同样问题，折腾半天没有解决，我是基于lean编译的

kenzok8, 2020-09-08:
readd support SS && Socks 已添加

xzjue5, 2020-09-08:
> readd support SS && Socks已添加

( cd /home/xzj/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1 ; ./tools/import-upstream.sh ; . ./init_env.sh "x86_64" "x86" "" "" "/home/xzj/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl" "/home/xzj/lede/dl"; export naive_flags="${naive_flags} ${naive_ccache_flags}" ; export OP_STAGING_DIR="/home/xzj/lede/staging_dir/target-x86_64_musl" ; mkdir -p out ; ./gn/out/gn gen "out/Release" --args="${naive_flags}" --script-executable="/home/xzj/lede/staging_dir/hostpkg/bin/python3" ; /home/xzj/lede/staging_dir/hostpkg/bin/ninja -C "out/Release" naive ; )
+ tr \n .
+ cut -d. -f1-4
+ cut -d= -f2 src/chrome/VERSION
+ have_version=85.0.4183.83
+ cat CHROMIUM_VERSION
+ want_version=85.0.4183.83
+ [ 85.0.4183.83 = 85.0.4183.83 ]
+ exit 0
./init_env.sh: line 30: curl: command not found
./init_env.sh: line 31: /home/xzj/lede/dl/chromeos-chrome-amd64-85.0.4176.0_rc-r1-merged.afdo.bz2: No such file or directory
./init_env.sh: line 38: curl: command not found
tar (child): /home/xzj/lede/dl/clang-n356902-4e813bbd-1.tgz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
./init_env.sh: line 46: curl: command not found
unzip:  cannot find or open /home/xzj/lede/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip, /home/xzj/lede/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip.zip or /home/xzj/lede/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip.ZIP.
bash: ./gn/out/gn: No such file or directory
ninja: Entering directory `out/Release'
ninja: fatal: chdir to 'out/Release' - No such file or directory
make[3]: *** [Makefile:88: /home/xzj/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1/.built] Error 1
make[3]: Leaving directory '/home/xzj/lede/feeds/small/naiveproxy'
time: package/feeds/small/naiveproxy/compile#0.78#3.52#8.33
make[2]: *** [package/Makefile:113: package/feeds/small/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/xzj/lede'
make[1]: *** [package/Makefile:107: /home/xzj/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xzj/lede'
make: *** [/home/xzj/lede/include/toplevel.mk:224：world] 错误 2

还是报错

lenngennd, 2020-09-09:
相同的报错+1

brianzc, 2020-09-09:
相同的报错+1

hgq5, 2020-09-09:
今天已经解决了，谢谢。

xzjue5, 2020-09-09:
> 今天已经解决了，谢谢。
怎么解决的

closed 2020-09-09 by zhucaidan

Itxiaoniao, 2020-09-13:
naiveproxy里面Makefile文件，文件名写错了，修改一下就可以了，.tar.gz 改成 .tar.xz

PKG_NAME:=naiveproxy
PKG_VERSION:=85.0.4183.83-3
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz

hzjnet, 2020-09-16:
如何解决的说一下吧

kenzok8, 2020-09-16:
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a

运行下此命令，重新拉取最新源