kenzok8/openwrt-packages #57: 编译失败 附错误代码

lenngennd, 2020-09-09:
( cd /home/mythos/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1 ; ./tools/import-upstream.sh ; . ./init_env.sh "x86_64" "x86" "" "" "/home/mythos/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl" "/home/mythos/lede/dl"; export naive_flags="${naive_flags} ${naive_ccache_flags}" ; export OP_STAGING_DIR="/home/mythos/lede/staging_dir/target-x86_64_musl" ; mkdir -p out ; ./gn/out/gn gen "out/Release" --args="${naive_flags}" --script-executable="/home/mythos/lede/staging_dir/hostpkg/bin/python3" ; /home/mythos/lede/staging_dir/hostpkg/bin/ninja -C "out/Release" naive ; )
+ cut -d= -f2 src/chrome/VERSION
+ tr \n .
+ cut -d. -f1-4
+ have_version=85.0.4183.83
+ cat CHROMIUM_VERSION
+ want_version=85.0.4183.83
+ [ 85.0.4183.83 = 85.0.4183.83 ]
+ exit 0
./init_env.sh: line 38: curl: command not found
tar (child): /home/mythos/lede/dl/clang-n356902-4e813bbd-1.tgz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
./init_env.sh: line 46: curl: command not found
unzip:  cannot find or open /home/mythos/lede/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip, /home/mythos/lede/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip.zip or /home/mythos/lede/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip.ZIP.
bash: ./gn/out/gn: No such file or directory
ninja: Entering directory `out/Release'
ninja: fatal: chdir to 'out/Release' - No such file or directory
Makefile:85: recipe for target '/home/mythos/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1/.built' failed
make[3]: *** [/home/mythos/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1/.built] Error 1
make[3]: Leaving directory '/home/mythos/lede/feeds/small/naiveproxy'
time: package/feeds/small/naiveproxy/compile#0.28#0.24#0.78
package/Makefile:111: recipe for target 'package/feeds/small/naiveproxy/compile' failed
make[2]: *** [package/feeds/small/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/mythos/lede'
package/Makefile:107: recipe for target '/home/mythos/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/mythos/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mythos/lede'
/home/mythos/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

richaradlee, 2020-09-09:
临时解决办法，已试可以完成云编译。

Add "src-git lienol https://github.com/Lienol/openwrt-package" to feeds.conf.default.

lenngennd, 2020-09-10:
> 临时解决办法，已试可以完成云编译。
> 
> Add "src-git lienol https://github.com/Lienol/openwrt-package" to feeds.conf.default.

问题依旧

lenngennd, 2020-09-11:
naiveproxy插件需要好梯子，ubuntu安装curl

closed 2020-09-11 by lenngennd

hzjnet, 2020-09-16:
下载后还是编译2不成功，提示如下：
bash: ./gn/out/gn: No such file or directory
bash: /openwrt/openwrt-sou-down-17.01.6/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.20_eabi/host/bin/ninja: No such file or directory
make[3]: *** [/openwrt/openwrt-sou-down-17.01.6/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.20_eabi/naiveproxy-85.0.4183.83-1/.built] Error 127
make[3]: Leaving directory `/openwrt/app/17016d0912/lienol/package/naiveproxy'
我的报错前面是一堆正常的gn/out/aaa行，明显下载的gn能自动正常解压：
( cd /openwrt/openwrt-sou-down-17.01.6/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.20_eabi/naiveproxy-85.0.4183.83-1 ; ./tools/import-upstream.sh ; . ./init_env.sh "arm" "ipq806x" "cortex-a15" "neon-vfpv4" "/openwrt/openwrt-sou-down-17.01.6/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl-1.1.20_eabi" "/openwrt/openwrt-sou-down-17.01.6/dl"; export naive_flags="${naive_flags} ${naive_ccache_flags}" ; export OP_STAGING_DIR="/openwrt/openwrt-sou-down-17.01.6/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.20_eabi" ; mkdir -p out ; ./gn/out/gn gen "out/Release" --args="${naive_flags}" --script-executable="/openwrt/openwrt-sou-down-17.01.6/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.20_eabi/host/bin/python3" ; /openwrt/openwrt-sou-down-17.01.6/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.20_eabi/host/bin/ninja -C "out/Release" naive ; )
Archive: /openwrt/openwrt-sou-down-17.01.6/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip.ZIP
extracting: gn/out/aaa/.clang-format
inflating: gn/out/aaa/.editorconfig
inflating: gn/out/aaa/.gitignore
inflating: gn/out/aaa/.style.yapf
inflating: gn/out/aaa/AUTHORS
inflating: gn/out/aaa/LICENSE
inflating: gn/out/aaa/OWNERS
.........
inflating: gn/out/aaa/src/util/worker_pool.h
inflating: gn/out/aaa/tools/find_unreachable.py
make[3]: Leaving directory /openwrt/app/17016d0912/lienol/package/naiveproxy' make[2]: Leaving directory /openwrt/openwrt-sou-down-17.01.6'
make[1]: Leaving directory `/openwrt/openwrt-sou-down-17.01.6'