kenzok8/openwrt-packages #58: 这是什么情况

wwz09, 2020-09-09:
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'https-dns-proxy', which does not exist
WARNING: Makefile 'package/feeds/kenzo/naiveproxy/Makefile' has a dependency on 'libnss', which does not exist

richaradlee, 2020-09-09:
临时解决办法，已试可以完成云编译。

Add "src-git lienol https://github.com/Lienol/openwrt-package" to feeds.conf.default.

harry3633, 2021-02-19:
我的云编译出现了这么多 ，这是为何：

```
WARNING: Makefile 'package/lean/luci-app-mwan3helper/Makefile' has a dependency on 'pdnsd-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'ipt2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'pdnsd-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'shadowsocksr-libev-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'shadowsocksr-libev-ssr-local', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'shadowsocksr-libev-server', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'trojan-plus', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'trojan-go', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'brook', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'chinadns-ng', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'dns2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'v2ray-plugin', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'simple-obfs', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'pdnsd-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'dns2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-local', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'simple-obfs', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'v2ray-plugin', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'trojan', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'ipt2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-server', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'shadowsocksr-libev-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'pdnsd-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'shadowsocksr-libev-ssr-local', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'simple-obfs', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'v2ray-plugin', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'trojan', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'ipt2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'shadowsocksr-libev-server', which does not exist
```
求助，谢谢！

closed 2023-02-01 by kenzok8