kenzok8/openwrt-packages #59: 编译出错

brianzc, 2020-09-09:
Makefile:85: recipe for target '/home/zc/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1/.built' failed
make[3]: *** [/home/zc/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1/.built] Error 1
make[3]: Leaving directory '/home/zc/lede/feeds/small/naiveproxy'
time: package/feeds/small/naiveproxy/compile#0.12#0.02#0.15
package/Makefile:111: recipe for target 'package/feeds/small/naiveproxy/compile' failed
make[2]: *** [package/feeds/small/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/zc/lede'
package/Makefile:107: recipe for target '/home/zc/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zc/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zc/lede'
/home/zc/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

lenngennd, 2020-09-11:
检查ubuntu是否安装curl  搭好梯子 naiveproxy下载很慢

myk4727633, 2020-10-13:
相同出错的问题，采用的是github云编译

tefiszx, 2020-10-14:
我也遇到了同样的问题，naive编译出错

> make[3]: Entering directory '/workdir/openwrt/feeds/kenzo/naiveproxy'
touch /workdir/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1/.prepared_a255e230b35d4fa48cf60921c3d84319_6664517399ebbbc92a37c5bb081b5c53_check
. /workdir/openwrt/include/shell.sh; gzip -dc /workdir/openwrt/dl/naiveproxy-86.0.4240.75-1.tar.gz | tar -C /workdir/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /workdir/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1

Applying ./patches/001-Remove-concurrency-limit.patch using plaintext: 
patching file src/net/tools/naive/naive_proxy.cc
patching file src/net/tools/naive/naive_proxy_bin.cc

Applying ./patches/100-build-add-OpenWrt-staging-libraries.patch using plaintext: 
patching file src/build/config/posix/BUILD.gn
Hunk #1 FAILED at 28.
1 out of 2 hunks FAILED -- saving rejects to file src/build/config/posix/BUILD.gn.rej
Patch failed!  Please fix ./patches/100-build-add-OpenWrt-staging-libraries.patch!
Makefile:86: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1/.prepared_a255e230b35d4fa48cf60921c3d84319_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1/.prepared_a255e230b35d4fa48cf60921c3d84319_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/kenzo/naiveproxy'
time: package/feeds/kenzo/naiveproxy/compile#2.53#1.76#4.04
make[2]: *** [package/feeds/kenzo/naiveproxy/compile] Error 2
package/Makefile:111: recipe for target 'package/feeds/kenzo/naiveproxy/compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.

tefiszx, 2020-10-15:
已解决
把两个patch文件修改一下即可。云编译的，在自定义脚本中增加两个命令
# repatch kenso8's naiveproxy patchfiles
sed -i 's/if (is_linux) {/if (is_linux || is_chromeos) {/' package/feeds/kenzo/naiveproxy/patches/100-build-add-OpenWrt-staging-libraries.patch
sed -i 's/@@ -746,7 +746,6/@@ -751,7 +751,6/' package/feeds/kenzo/naiveproxy/patches/101-build-drop-useless-deps-simd_asm.patch

kenzok8, 2020-10-15:
已经修复错误，重新拉取源码就可以了！也谢谢你细致的解答

closed 2023-02-01 by kenzok8