kenzok8/openwrt-packages #70: 编译naive会遇到补丁出错以及ncurses版本过低的问题

tefiszx, 2020-10-15:
使用K神的包，很方便，所有的常用软件都一站打包配齐了。然而在编译的时候，naiveproxy那个包总是出问题。naive的上游开发者也提醒过交叉编译的问题，建议最好用他的release提供的二进制文件。
我这两天编译了k神的包，发现naive编译出错主要是两个问题。
1、是patch目录里的两个补丁包100-build-add-OpenWrt-staging-libraries.patch 和 101-build-drop-useless-deps-simd_asm.patch没有同步更新，导致上游开发者更新了代码后，打补丁出错的问题。
2、curses版本依赖过低，现在的curses基本上都是6版本，不知道哪个makefile里依然是对5的依赖。造成编译的时候找不到libtinfo.so.5这个库文件

只要解决了这个两个问题。编译就正常了。两个补丁包的修改方法可以区参考其他repositroy的同名称文件。k神最后一次更新的是在9天前，补丁包已经outdate了。希望k神及时更新，以飨网友

hathlife, 2020-10-15:
```
WARNING: kmod-nf-conntrack6 is not available in the kernel config - generating empty package
make[3]: *** [Makefile:93: /home/toonaive/openwrt/build_dir/target-mipsel_24kc_musl/naiveproxy-86.0.4240.75-1/.prepared_25acb5986777f484ce784f87b5713ad5_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[2]: *** [package/Makefile:113: package/feeds/kenzo/naiveproxy/compile] Error 2
make[1]: *** [package/Makefile:107: /home/toonaive/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
```
是不是遇到这种问题啊？很困扰啊

tefiszx, 2020-10-15:
> ```
> WARNING: kmod-nf-conntrack6 is not available in the kernel config - generating empty package
> make[3]: *** [Makefile:93: [...]
我的openwrt架构和你的不太一样，编译出错的提示也不同。我的编译出错提示是打补丁错误，看你的好像是依赖出了问题。

wmjlxl, 2020-10-15:
我刚才 云编译也出错了。求方法。

Hunk #1 FAILED at 28.
1 out of 2 hunks FAILED -- saving rejects to file src/build/config/posix/BUILD.gn.rej
Patch failed!  Please fix ./patches/100-build-add-OpenWrt-staging-libraries.patch!
Makefile:86: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1/.prepared_75f65b3129567b78394b83a3a76880d2_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: Leaving directory '/workdir/openwrt/package/small/naiveproxy'
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1/.prepared_75f65b3129567b78394b83a3a76880d2_6664517399ebbbc92a37c5bb081b5c53] Error 1
time: package/small/naiveproxy/compile#2.44#1.71#3.73
package/Makefile:111: recipe for target '
make[2]: *** [package/small/naiveproxy/compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

wmjlxl, 2020-10-16:
> > 我刚才 云编译也出错了。求方法。
> > Hunk #1 FAILED at 28.
> > 1 out of 2 hunks FAILED -- saving rejects to file src/build/config/posix/BUILD.gn.rej
> > Patch failed! Please fix ./patches/100-build-add-OpenWrt-staging-libraries.patch!
> > Makefile:86: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1/.prepared_75f65b3129567b78394b83a3a76880d2_6664517399ebbbc92a37c5bb081b5c53' failed
> > make[3]: Leaving directory '/workdir/openwrt/package/small/naiveproxy'
> > make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1/.prepared_75f65b3129567b78394b83a3a76880d2_6664517399ebbbc92a37c5bb081b5c53] Error 1
> > time: package/small/naiveproxy/compile#2.44#1.71#3.73
> > package/Makefile:111: recipe for target '
> > make[2]: *** [package/small/naiveproxy/compile] Error 2
> > make[2]: Leaving directory '/workdir/openwrt'
> > make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
> > package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
> > make[1]: Leaving directory '/workdir/openwrt'
> > /workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
> > make: *** [world] Error 2
> > 看这个issue中我的跟帖
> > [#59 (comment)](https://github.com/kenzok8/openwrt-packages/issues/59#issuecomment-708892892)

明白了，非常感谢。

closed 2020-10-30 by tefiszx