kenzok8/openwrt-packages #71: 有关在原版19.07分支下编译passwall的问题

hathlife, 2020-10-16:
### **1. kcptun**
找不到依赖是因为官方代码库自带了一个kcptun-c，不过Makefile里面定义的版本很老了，最简单的方法就是menuconfig里面自己取消passwall对kcptun-client的依赖，，然后单独编译官方库里面的kcptun-c，有必要的话手工修改Makefile里面的版本号和代码包hash。不知道可不可以在本仓库里面用条件判断解决这个问题

### **2.Golang版本**
官方19.07 feed源里面的Golang工具链，Makefile里面定义的版本只到1.13，无法编译small源里面的v2ray*包。解决方法同1，先修改官方feed/package/golang/golang-version.mk里面的定义到最新版本：
```
GO_VERSION_MAJOR_MINOR:=1.15
GO_VERSION_PATCH:=2
```
再次make V=99 从错误信息中得到hash填入feed/package/golang/golang/Makefile替换即可。

### **3.upx压缩**
v2ray需要的upx压缩工具包可从lean库的tools文件夹中获得，见#31

### **4.luci-lib-ipkg**
passwall界面运行需要luci-lib-ipkg，但是Makefile依赖里面没有。如果是构建软件包模块的话，安装后出现luci界面打不开的情况命令行手动输入opkg install luci-lib-ipkg即可

### **5.passwall工作效率问题**
这样编译出来的passwall软件包组合 安装在newifi d2官方19.07.4环境下使用v2ray模式，在cpu占用底下的情况下，同节点比以前其他人自编译的固件中的老版passwall套件速度慢很多，延迟很高，Google连接检测延迟没下过4位数（节点列表ping30ms），YouTube缓冲速度也比别人的老整合固件慢很多，没法像别人一样跑满。调节dns过滤模式在重启似乎可以得到暂时改善。不知有没有办法可以排查错误？

kenzok8, 2020-10-22:
这个吧，在我的能力范围解决不了，要问Lienol作者

hathlife, 2020-10-22:
解决了，开启chinadns-ng明显改善速度

closed 2020-10-22 by hathlife

matterwind, 2021-06-16:
终于找到缺少kcptun-client依赖的靠谱解决办法了，和我怀疑的一样，和官方包冲突。但是，已经在menuconfig里面取消passwall的kcotun选项了，还是失败。楼主能再详细说下怎么处理吗？没太看懂，怎么修改makefile，修改passwall的makefile？把包名改掉？可以干掉官方的kcptun-c吗？谢谢啦

matterwind, 2021-06-16:
能否直接用这里kcptun的makefile替换掉openwrt包里面的？

matterwind, 2021-06-18:
> ### **1. kcptun**
> 找不到依赖是因为官方代码库自带了一个kcptun-c，不过Makefile里面定义的版本很老了，最简单的方法就是menuconfig里面自己取消passwall对kcptun-client的依赖，，然后单独编译官方库里面的kcptun-c，有必要的话手工修改Makefile里面的版本号和代码包hash。不知道可不可以在本仓库里面用条件判断解决这个问题
> 
> ### **2.Golang版本**
> 官方19.07 feed源里面的Golang工具链，Makefile里面定义的版本只到1.13，无法编译small源里面的v2ray*包。解决方法同1，先修改官方feed/package/golang/golang-version.mk里面的定义到最新版本：
> 
> ```
> GO_VERSION_MAJOR_MINOR:=1.15
> GO_VERSION_PATCH:=2
> ```
> 
> 再次make V=99 从错误信息中得到hash填入feed/package/golang/golang/Makefile替换即可。
> 
> ### **3.upx压缩**
> v2ray需要的upx压缩工具包可从lean库的tools文件夹中获得，见#31
> 
> ### **4.luci-lib-ipkg**
> passwall界面运行需要luci-lib-ipkg，但是Makefile依赖里面没有。如果是构建软件包模块的话，安装后出现luci界面打不开的情况命令行手动输入opkg install luci-lib-ipkg即可
> 
> ### **5.passwall工作效率问题**
> 这样编译出来的passwall软件包组合 安装在newifi d2官方19.07.4环境下使用v2ray模式，在cpu占用底下的情况下，同节点比以前其他人自编译的固件中的老版passwall套件速度慢很多，延迟很高，Google连接检测延迟没下过4位数（节点列表ping30ms），YouTube缓冲速度也比别人的老整合固件慢很多，没法像别人一样跑满。调节dns过滤模式在重启似乎可以得到暂时改善。不知有没有办法可以排查错误？
> 附Youtube测试图
> 
> 电脑客户端
> [image]
> 路由模式，刚刚启动 v2ray版本 4.30.0
> [image]
> 路由模式，长时间工作后：
> [image]
> 
> 不知道该从哪里排查问题 买路由的时候卖家配的固件里面的老版passwall套件（五月份的）就很快。

求大佬解答！