kenzok8/openwrt-packages #82: 不懂就问 AdGuardHome和luci-app-adguardhome 有什么区别？？？

gbaoye, 2020-11-07:
两个都编译会产生什么问题？

Carrjiang, 2020-11-07:
AdguardHome是luci-app-adguardhome的功能功能执行文件，也就是原版的AdGuardHome，编译了前者在第一次使用时不需要下载执行文件，luci-app-adguardhome只是AdGuardHome的OpenWrt控制面板

gbaoye, 2020-11-07:
> AdguardHome是luci-app-adguardhome的功能功能执行文件，也就是原版的AdGuardHome，编译了前者在第一次使用时不需要下载执行文件，luci-app-adguardhome只是AdGuardHome的OpenWrt控制面板
> […](#)
虽然我没有听明白，但是非常感谢你！谢谢你！！！

yege16888, 2021-03-22:
通俗的说  luci-app-adg是软件的UI界面 adg则是执行文件。

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.