kenzok8/openwrt-packages #86: ssr-plus 运行时，OpenConnect VPN用户vpns+无法透明科学上网

dyinandy, 2020-11-16:
ssr-plus正常运行，无论是openconnect server设置与LAN同一网段还是不同的网段，vpns+客户端无法透明科学上网。

原因：
uci -p /var/state get network.lan.ifname 获得是LAN的eth(x). 
之前为什么没发生问题是因为：uci get -P/var/state network.lan.ifname 命令无返回。

表: NAT
链 PREROUTING
数据包 	流量 	对象 	协议 	入口 	出口 	源地址 	目标地址 	选项
0 	0.00 B 	SS_SPEC_WAN_AC 	tcp 	eth(x) 	* 	0.0.0.0/0 	0.0.0.0/0 	/* _SS_SPEC_RULE_ */

表: Mangle
链 PREROUTING 
数据包 	流量 	对象 	协议 	入口 	出口 	源地址 	目标地址 	选项
0 	0.00 B 	SS_SPEC_TPROXY 	udp 	eth(x) 	* 	0.0.0.0/0 	0.0.0.0/0 	/* _SS_SPEC_RULE_ */

若只eth(x)就会有导致只能LAN下面用户可以透明出关。

临时解决方法：
`删除 IFNAME=$(uci -P /var/state get network.lan.ifname 2>/dev/null)`
`以及后面的两处：${IFNAME:+-i $IFNAME}`

结果：
表: NAT
链 PREROUTING
数据包 	流量 	对象 	协议 	入口 	出口 	源地址 	目标地址 	选项
62 	3.15 KB 	SS_SPEC_WAN_AC 	tcp 	* 	* 	0.0.0.0/0 	0.0.0.0/0 	/* _SS_SPEC_RULE_ */

表: Mangle
链 PREROUTING
数据包 	流量 	对象 	协议 	入口 	出口 	源地址 	目标地址 	选项
645 	54.60 KB 	SS_SPEC_TPROXY 	udp 	* 	* 	0.0.0.0/0 	0.0.0.0/0 	/* _SS_SPEC_RULE_ */

vpns+ 用户科学上网正常。

closed 2021-01-14 by kenzok8

xizhilin, 2021-02-26:
已经安装了Openconnect，能够访问内网，无法科学上网。
根据你的文章，已经在ssr-rules找到：
local IFNAME=$(uci -P /var/state get network.$name.ifname 2>/dev/null)
		[ -n "$IFNAME" ] && $IPT -I PREROUTING 1 ${IFNAME:+-i $IFNAME} -p tcp $EXT_ARGS $MATCH_SET -m comment --comment "$TA

请问，IFNAME前面的local要删除么？
谢谢你的答复。

xizhilin, 2021-02-27:
你好，按教程删除SSR-rule后，SSR无法运行。大侠，有没有找到其他的解决方法了？

fzp0723, 2022-05-10:
@xizhilin 找上面说的那两行把 ${IFNAME:+-i $IFNAME} 这部分删除了就行, 总共两个tcp 一个 udp一个