kenzok8/openwrt-packages #105: 今天20:10分拉取，xray不见了

guochan2019, 2021-01-25:
Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
Already up to date.
Create index file './feeds/packages.index'
Updating feed 'luci' from 'https://github.com/coolsnowwolf/luci' ...
Already up to date.
Create index file './feeds/luci.index'
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git' ...
Already up to date.
Create index file './feeds/routing.index'
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
Already up to date.
Create index file './feeds/telephony.index'
Updating feed 'freifunk' from 'https://github.com/freifunk/openwrt-packages.git' ...
Already up to date.
Create index file './feeds/freifunk.index'
Updating feed 'kenzo' from 'https://github.com/kenzok8/openwrt-packages' ...
Already up to date.
Create index file './feeds/kenzo.index'
Updating feed 'small' from 'https://github.com/kenzok8/small' ...
Already up to date.
Create index file './feeds/small.index'
**WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'xray', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'xray', which does not exist**
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed freifunk.
Installing all packages from feed kenzo.
Installing all packages from feed small.

Netflixxp, 2021-01-25:
一样

kenzok8, 2021-01-25:
openwrt-packages仓库里本来就没放入xray，另外small里有，现在helloworld仓库地址有xray，passwall仓库地址也有

Netflixxp, 2021-01-25:
就是这波大佬更新后就会出现，咋办呢
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'xray', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssrplus/Makefile' has a dependency on 'xray', which does not exist

kenzok8, 2021-01-25:
rm -rf ./feeds
./scripts/feeds clean
./scripts/feeds update -a

yuyongji1987, 2021-01-26:
> rm -rf ./feeds
> ./scripts/feeds clean
> ./scripts/feeds update -a

仍然存在这个warning

yuyongji1987, 2021-01-26:
手动把small下面的xray链接到kenzo下面，可以解决

Netflixxp, 2021-01-26:
请教，是在./feeds里面加入下面这行代码吗？

`svn co https://github.com/kenzok8/small/tree/master/xray`

yuyongji1987, 2021-01-26:
> 请教，是在./feeds里面加入下面这行代码吗？
> 
> `svn co https://github.com/kenzok8/small/tree/master/xray`

cp -R ./feeds/small/xray ./feeds/kenzo
./scripts/feeds update -a && ./scripts/feeds install -a

guochan2019, 2021-01-27:
> openwrt-packages仓库里本来就没放入xray，另外small里有，现在helloworld仓库地址有xray，passwall仓库地址也有

升级前都是正常编译的，拉取新代码后就这样了，今天按上面说的几个方法再次拉取，还是不行，麻烦大佬回看下哪里出问题了

guochan2019, 2021-01-27:
> > 请教，是在./feeds里面加入下面这行代码吗？
> > `svn co https://github.com/kenzok8/small/tree/master/xray`
> 
> cp -R ./feeds/small/xray ./feeds/kenzo
> ./scripts/feeds update -a && ./scripts/feeds install -a

兄弟这个方法也不行

kenzok8, 2021-01-28:
我特地去编译了一次，没错误，你实在不行就make distclean

kenzo wu

Netflixxp, 2021-01-28:
> 我特地去编译了一次，没错误，你实在不行就make distclean kenzo wu
> […](#)

是的，make distclean后没问题了

guochan2019, 2021-01-28:
> 我特地去编译了一次，没错误，你实在不行就make distclean kenzo wu
> […](#)

感谢，make distclean后解决

closed 2021-01-30 by kenzok8