kenzok8/openwrt-packages #107: 你好，麻烦看下凌晨更新的gost编译出错了

hibuddies, 2021-01-31:
` * check_data_file_clashes: Package luci-app-gost wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/config/gost
	But that file is already provided by package  * gost
 * check_data_file_clashes: Package luci-app-gost wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/init.d/gost
	But that file is already provided by package  * gost
 * opkg_install_cmd: Cannot install package luci-app-gost.
 * check_data_file_clashes: Package luci-app-gost wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/config/gost
	But that file is already provided by package  * gost
 * check_data_file_clashes: Package luci-app-gost wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/init.d/gost
	But that file is already provided by package  * gost
 * opkg_install_cmd: Cannot install package luci-i18n-gost-zh-cn.
 * resolve_conffiles: Existing conffile /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/ppp/options-opkg.
make[2]: *** [package/install] Error 255
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make: *** [world] Error 2
package/Makefile:65: recipe for target 'package/install' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:108: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed`

closed 2021-02-04 by kenzok8

aceslup, 2021-02-19:
我最近是这个问题，有解决方案了吗，就结帖了？ @kenzok8 @hibuddies

hibuddies, 2021-02-19:
> 我最近是这个问题，有解决方案了吗，就结帖了？ @kenzok8 @hibuddies

我暂时换别的了

aceslup, 2021-02-26:
我修改gost中的Makefile，/etc/init.d/gost 和 /etc/config/gost 给重新命名，好似能正常编译。不知道他人还有没有问题？