kenzok8/openwrt-packages #110: 额 更新软件包 编译出错,距离上次软件包8天左右

80houco, 2021-02-04:
`Collected errors:
 * check_data_file_clashes: Package xray-core wants to install file /home/lh/lede/build_dir/target-x86_64_musl/root-x86/usr/bin/v2ray
        But that file is already provided by package  * v2ray
 * opkg_install_cmd: Cannot install package luci-app-passwall.
 * check_data_file_clashes: Package xray-core wants to install file /home/lh/lede/build_dir/target-x86_64_musl/root-x86/usr/bin/v2ray
        But that file is already provided by package  * v2ray
 * opkg_install_cmd: Cannot install package xray-core.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/lh/lede'
package/Makefile:108: recipe for target '/home/lh/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/lh/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/lh/lede'
/home/lh/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2`

kenzok8, 2021-02-04:
./scripts/feeds clean后重新拉取下吧，这几天更新频繁

80houco, 2021-02-04:
> ./scripts/feeds clean后重新拉取下吧，这几天更新交替

OK'thanks for the reply

closed 2021-02-24 by kenzok8