kenzok8/openwrt-packages #111: 每次都是xray报错

Gx3dong, 2021-02-04:
Makefile:106: recipe for target '/home/gx3dong/openwrt/build_dir/target-x86_64_musl/Xray-core-1.2.4/.built' failed
make[3]: *** [/home/gx3dong/openwrt/build_dir/target-x86_64_musl/Xray-core-1.2.4/.built] Error 2
make[3]: Leaving directory '/home/gx3dong/openwrt/feeds/small/xray-core'
time: package/feeds/small/xray-core/compile#2.99#1.86#20.19
package/Makefile:111: recipe for target 'package/feeds/small/xray-core/compile' failed
make[2]: *** [package/feeds/small/xray-core/compile] Error 2
make[2]: Leaving directory '/home/gx3dong/openwrt'
package/Makefile:107: recipe for target '/home/gx3dong/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/gx3dong/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gx3dong/openwrt'
/home/gx3dong/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

Gx3dong, 2021-02-05:
官方固件就是编译不进去  求大神指点下

Gx3dong, 2021-02-05:
github已经是拉取最新的了

kenzok8, 2021-02-05:
什么源码？lean的还是Lienol的？
./scripts/feeds update -a
rm -rf /feeds/small/v2ray
rm -rf /feeds/small/v2ray-plugin
rm -rf /feeds/small/xray-core
./scripts/feeds install -a

kenzok8, 2021-02-05:
是官网的源码吗？还是coolsnowwolf的？如果是后者，删除源码里的v2ray
rm -rf package/lean/v2ray
rm -rf package/lean/v2ray-plugin

Gx3dong, 2021-02-05:
我用的官方的源码   ，官方源码要在master编译  还是v19.07.6分支编译，

Gx3dong, 2021-02-05:
那你的依赖库里面的v2ray 和v2ray-plugin要删除吗？    谢谢

closed 2021-02-09 by kenzok8