kenzok8/openwrt-packages #115: Error compiling with Github-Action

M33FT, 2021-02-06:
Boss, can you explain to me how I can connect your Repo to Coolwoolf / lede and 
compile the ROM with Github-Action for RaspberryPi 4B?

I easily compiled CoolWolf / LEDE + xiaorouji / openwrt-passwall with the same Github-Action.
Now I have a config file. I fixed it and I know there is no case and now everything is from the file feeds.conf.default, the specifications of which are as follows:

src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git packages https://github.com/coolsnowwolf/packages
src-git luci https://github.com/coolsnowwolf/luci
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git
src-git freifunk https://github.com/freifunk/openwrt-packages.git
src-git helloworld https://github.com/fw876/helloworld

Now I get an error when I get output after a while.

Thank you for guiding me.

kenzok8, 2021-02-06:
Delete v2ray and v2ray-plugin from Coolwoolf/LEDE
rm -rf package/lean/v2ray && rm -rf package/lean/v2ray-plugin

kenzok8, 2021-02-06:
Delete v2ray and v2ray-plugin from Coolwoolf/LEDE
rm -rf package/lean/v2ray && rm -rf package/lean/v2ray-plugin
  

Or delete v2ray, v2ray-plugin in small

closed 2021-02-08 by kenzok8

M33FT, 2021-02-09:
@kenzok8
I wanted to compile the ROM in Github-Action, I did as you did and deleted those 
two folders and gave my Repo address for Github-Action, but in the end it gives an error and does not run.
Can you guide me?

WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'antfs-mount', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'wsdd2', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/frp/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libgstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1-libs', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/kcptun/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/libs/libselinux/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'alsa-utils', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'shairport-sync-openssl', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifs-mount/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a dependency on 'ksmbd-server', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
[... cut ...]
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-program_options', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-date_time', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusicGo/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist

kenzok8, 2021-02-09:
https://github.com/kenzok8/LEDE-x86_64

M33FT, 2021-02-09:
@kenzok8

Thanks boss for your link. I saw them. It was so good.
But you can give me more guidance on how and where I can start learning so that I can create Config files for compilation in
 Github-Action myself.
I know the config file. How do I make it, but what you made is complicated in my opinion.
Your device is probably NanoPi_R2S, but the devices I want to compile are RaspberryPI4, RaspberryPi3B, WRT1900ACS, WRT1200,
 WRT32X.

Thanks Boss ;-)

M33FT, 2021-02-10:
@kenzok8

[image]