kenzok8/openwrt-packages #116: ./scripts/feeds install -a的时候，老是提示缺少kcptun-client

panlivictory, 2021-02-07:
已在feeds.conf.default里面添加了
src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git small https://github.com/kenzok8/small

但./scripts/feeds install -a的时候，老是提示
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'kcptun-client', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'kcptun-client', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'kcptun-client', which does not exist

请问怎么解决了。谢谢了

MrMatttw, 2021-02-08:
同样的问题，官方原版openwrt19.07也在这一步报warnning

panlivictory, 2021-02-08:
> 同样的问题，官方原版openwrt19.07也在这一步报warnning

是的。我也是官方原版19.07编译的时候在这报错

kenzok8, 2021-02-08:
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a

MrMatttw, 2021-02-08:
问题依然存在

kenzok8, 2021-02-08:
有些错误也能成功编译，如果编译失败，建议运行make distclean
因为我云编译，本地编译都ok,你的错误是个例

closed 2021-02-08 by kenzok8

panlivictory, 2021-02-08:
> 问题依然存在

我成功了。刚开始按K大给的方法，也没有成功。后来我改变了一下，提供给你参考
1、先./scripts/feeds clean
1、在feeds.conf.default先不要添加K大的两个源码链接，完全保持官网OPENWRT 19.07的原样，走一遍
./scripts/feeds update -a
./scripts/feeds install -a

这个时候，你会发现全有一些错误提示，告知有一些依赖没有，好像是通病，这个时候再跑一次
./scripts/feeds update -a
./scripts/feeds install -a
你会发现错误提示没有了，全正常

3、再在feeds.conf.default添加K大的两个源码链接，跑一次
./scripts/feeds update -a
./scripts/feeds install -a

这个时候就全正常了。

panlivictory, 2021-02-08:
> 我成功了。刚开始按K大给的方法，也没有成功。后来我改变了一下，提供给你参考
> 1、先./scripts/feeds clean
> 1、在feeds.conf.default先不要添加K大的两个源码链接，完全保持官网OPENWRT 19.07的原样，走一遍
> ./scripts/feeds update -a
> ./scripts/feeds install -a [...]

晕，发现这里正常后，我make menuconfig的时候，还是一样出错。也是提示
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'kcptun-client', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'kcptun-client', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-vssr/Makefile' has a dependency on 'kcptun-client', which does not exist

kenzok8, 2021-02-08:
1、feeds.conf.default 注销掉small与 openwrt-packages. 运行./scripts/feeds clean
2、lede/package$下运行 或者openwrt/package$下运行
       git clone https://github.com/kenzok8/openwrt-packages.git
3、./scripts/feeds update -a
     ./scripts/feeds install -a
      make menuconfig 我只能帮到这里了，如果再出错，你烧支香吧

matterwind, 2021-06-16:
我也遇到了类似的问题，用的是友善之家官方固件，编译到最后总是失败，怀疑是不是passwall里面kcptun和原版里面的kuptun有冲突？