kenzok8/openwrt-packages #120: bypass报错Makefile' has a dependency on 'shadowsocksr-libev-ssr-redir',

Netflixxp, 2021-02-23:
```
WARNING: Makefile 'package/feeds/kenzo/luci-app-bypass/Makefile' has a dependency on 'shadowsocksr-libev-ssr-redir', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-bypass/Makefile' has a dependency on 'shadowsocksr-libev-ssr-server', which does not exist
```
make distclean也没用

kenzok8, 2021-02-24:
看下说明文档！

Netflixxp, 2021-02-24:
找到了，太靠下了，哈哈
```
find package/*/ feeds/*/ -maxdepth 2 -path "*luci-app-bypass/Makefile" | xargs -i sed -i 's/shadowsocksr-libev-ssr-redir/shadowsocksr-libev-alt/g' {}
find package/*/ feeds/*/ -maxdepth 2 -path "*luci-app-bypass/Makefile" | xargs -i sed -i 's/shadowsocksr-libev-ssr-server/shadowsocksr-libev-server/g' {}
```

closed 2021-02-25 by kenzok8