kenzok8/openwrt-packages #122: xray编译失败

123485k, 2021-02-24:
我使用了openwrt 19.07.7源码和您的feeds，在编译的过程中，无法编译xray，报错如下
```
github.com/xtls/xray-core/transport/internet/tls
github.com/xtls/go
# github.com/xtls/xray-core/transport/internet/tls
../../transport/internet/tls/config.go:323:21: undefined: tls.CipherSuites
note: module requires Go 1.15
github.com/lucas-clemente/quic-go/internal/qtls
# github.com/lucas-clemente/quic-go/internal/qtls
pkg/mod/github.com/lucas-clemente/quic-go@v0.19.3/internal/qtls/qtls.go:79:25: cann
ot convert cert (type *tls.Certificate) to type *qtls.Certificate
note: module requires Go 1.14
# github.com/xtls/go
pkg/mod/github.com/xtls/go@v0.0.0-20201118062508-3632bf3b7499/auth.go:29:7: undefin
ed: ecdsa.VerifyASN1
pkg/mod/github.com/xtls/go@v0.0.0-20201118062508-3632bf3b7499/key_schedule.go:177:1
6: xShared.FillBytes undefined (type *big.Int has no field or method FillBytes)

Cleaning module download cache (golang/go#27455)

make[3]: *** [Makefile:150: /home/lin/openwrt/build_dir/target-mipsel_24kc_musl/xra
y-core-1.3.0/.built] Error 2
make[3]: Leaving directory '/home/lin/openwrt/feeds/small/xray-core'
time: package/feeds/small/xray-core/compile#5.69#1.88#101.60
make[2]: *** [package/Makefile:113: package/feeds/small/xray-core/compile] Error 2
make[2]: Leaving directory '/home/lin/openwrt'
make[1]: *** [package/Makefile:107: /home/lin/openwrt/staging_dir/target-mipsel_24k
c_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lin/openwrt'
make: *** [/home/lin/openwrt/include/toplevel.mk:227: world] Error 2
```
请问要如何解决？

kenzok8, 2021-02-27:
最新更新频繁openwrt 19.07.7官方源码大概率会报错，建议是Lienol源码去编译，如果非得要用官方源码，建议不添加我的库先编译，成功了再添加，也知道问题出在哪

kenzok8, 2021-02-27:
至于xray内核，最近好像不会出错了

closed 2023-02-01 by kenzok8