kenzok8/openwrt-packages #125: 你好，小白有个问题，就是将ssrplus编译进lienol里？

cc-monkey, 2021-03-02:
将ssrplus编译进lienol里，在编译ssrplus方括号选项 不知道改怎么选。有一次全部选上了，最后编译出错。
[image]

kenzok8, 2021-03-07:
<*> luci-app-ssr-plus
[*] Include Shadowsocks
[ ] Include Shadowsocks V2ray Plugin
[*] Include Xray (V2RAY/Trojan-GO implemented)
[ ] Include Trojan (OpenSSL)
[ ] Include Redsocks2
[ ] Include NaiveProxy
[ ] Include Kcptun
[*] Include ShadowsocksR Server

kenzok8, 2021-03-07:
<*> luci-app-ssr-plus

[*] Include Shadowsocks

[*] Include Xray (V2RAY/Trojan-GO implemented)

[*] Include ShadowsocksR Server

jiahanbryant, 2021-04-19:
请问这一项，现在是被包含在第三项里了么？
[image]