kenzok8/openwrt-packages #126: 编译出现WARNING: Makefile 'package/feeds/kenzo/luci-app-adguardhome/Makefile' has a dependency on 'adguardhome', which does not exist

herunyu, 2021-03-03:
大佬，尝试编译adguard home进去的时候报这个，请教是否有影响？

hdgeek, 2021-03-03:
这个意思是adguardhome 文件不存在，编译时会报错，需要修改Makefile文件，将`LUCI_DEPENDS:=+ca-certs +curl +wget-ssl +PACKAGE_$(PKG_NAME)_INCLUDE_binary:adguardhome` 修改为 `LUCI_DEPENDS:=+ca-certs +curl +wget-ssl +PACKAGE_$(PKG_NAME)_INCLUDE_binary:AdGuardHome`。

另，作者修改了AdGuardHome运行位置，需要将AdGuardHome编译文件（Makefile)做进行修改（2处）。

第一处将`$(INSTALL_DIR) $(1)/usr/bin/AdGuardHome`修改为`$(INSTALL_DIR) $(1)/usr/bin`

第二处将`$(INSTALL_BIN) $(GO_PKG_BUILD_BIN_DIR)/AdGuardHome $(1)/usr/bin/AdGuardHome/AdGuardHome`，修改为`$(INSTALL_BIN) $(GO_PKG_BUILD_BIN_DIR)/AdGuardHome $(1)/usr/bin/AdGuardHome`

如果修改不来的，可下载我修改好的文件，进行覆盖（其中 AdGuardHome 不是编译版，而是官方的已经编译好的最新正式版），下载地址：[蓝奏云](https://wwx.lanzoui.com/iIuGrmfiv9a)

**由于更换了运行文件位置，更新时不要勾选保存配置文件**

herunyu, 2021-03-03:
> 这个意思是adguardhome 文件不存在，编译时会报错，需要修改Makefile文件，将`LUCI_DEPENDS:=+ca-certs +curl +wget-ssl +PACKAGE_$(PKG_NAME)_INCLUDE_binary:adguardhome` 修改为 `LUCI_DEPENDS:=+ca-certs +curl +wget-ssl +PACKAGE_$(PKG_NAME)_INCLUDE_binary:AdGuardHome`。
> 
> 补充，需要修改以下地方，否则，编译后插件依然是英文和更新固件配置文件丢失
> 
> 1、`/etc/config/AdGuardHome.yaml`改为`/etc/AdGuardHome.yaml`
> 2、将zh_Hans文件夹名改为zh-cn（适用于L大神源码）
> 
> 另，作者修改了AdGuardHome运行位置，需要将AdGuardHome编译文件（Makefile)做进行修改（2处）。
> 
> 第一处将`$(INSTALL_DIR) $(1)/usr/bin/AdGuardHome`修改为`$(INSTALL_DIR) $(1)/usr/bin`
> 
> 第二处将`$(INSTALL_BIN) $(GO_PKG_BUILD_BIN_DIR)/AdGuardHome $(1)/usr/bin/AdGuardHome/AdGuardHome`，修改为`$(INSTALL_BIN) $(GO_PKG_BUILD_BIN_DIR)/AdGuardHome $(1)/usr/bin/AdGuardHome`
> 
> 如果修改不来的，可下载我修改好的文件，进行覆盖（其中 AdGuardHome 不是编译版，而是官方的已经编译好的最新正式版），下载地址：[蓝奏云](https://wwx.lanzoui.com/iwolnmfs83g)
> 
> **由于更换了运行文件位置，更新时不要勾选保存配置文件**

感谢！如果之前没有装过adguard home就直接修改路径编译，仍然可以勾选保存配置文件吧?

closed 2021-08-09 by herunyu

cs-star, 2021-10-03:
WARNING: Makefile 'package/feeds/small/brook/Makefile' has a build dependency on 'upx/host', which does not exist
WARNING: Makefile 'package/feeds/small/hysteria/Makefile' has a build dependency on 'upx/host', which does not exist
WARNING: Makefile 'package/feeds/small/trojan-go/Makefile' has a build dependency on 'upx/host', which does not exist
WARNING: Makefile 'package/feeds/small/v2ray-core/Makefile' has a build dependency on 'upx/host', which does not exist
WARNING: Makefile 'package/feeds/small/v2ray-plugin/Makefile' has a build dependency on 'upx/host', which does not exist
WARNING: Makefile 'package/feeds/small/xray-core/Makefile' has a build dependency on 'upx/host', which does not exist
WARNING: Makefile 'package/feeds/small/xray-plugin/Makefile' has a build dependency on 'upx/host', which does not exist

zhangguanzhang, 2022-05-22:
```
WARNING: Makefile 'package/community/luci-app-adguardhome/Makefile' has a dependency on 'adguardhome', which does not exist
```
[adguardhome/Makefile最后一行的](https://github.com/kenzok8/openwrt-packages/blob/master/adguardhome/Makefile#L88) 
```
$(eval $(call BuildPackage,adguardhome))
```
至于上面俩老弟的upx/host是你没有host 的 upx 工具链，和adg 没有任何关系， upx可以试试：
```
svn export https://github.com/coolsnowwolf/lede/trunk/tools/upx   tools/upx
```

Hmilyqiumei, 2022-11-27:
WARNING: Makefile 'package/feeds/luci/luci-app-mosdns/Makefile' has a dependency on 'v2ray-geoip', which does not exist

Hmilyqiumei, 2022-11-27:
WARNING: Makefile 'package/feeds/luci/luci-app-mosdns/Makefile' has a dependency on 'v2ray-geoip', which does not exist

折腾2天了，有大佬给解决下不