kenzok8/openwrt-packages #130: 编译luci-app-ssr-plus时报错

Ginua8292, 2021-03-06:
* satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ssr-plus:
 * 	shadowsocksr-libev-ssr-redir
 * 	shadowsocksr-libev-ssr-server
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/ginua/lede'
make[1]: *** [package/Makefile:108: /home/ginua/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/ginua/lede'
make: *** [/home/ginua/lede/include/toplevel.mk:224：world] 错误 2
 
这二个依赖没有

Gabrielxzx, 2021-03-07:
> * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ssr-plus:
> * shadowsocksr-libev-ssr-redir
> * shadowsocksr-libev-ssr-server
> * opkg_install_cmd: Cannot install package [...]

我编译时也遇到了，去lean仓库看了看，不需要这两个依赖，不知道是从哪边更新的

kenzok8, 2021-03-07:
bypass ssr_plus这些默认配置编译，没啥问题的

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.