kenzok8/openwrt-packages #147: v2ray-plugin 编译失败

jiahanbryant, 2021-04-19:
Hi Guys,
今天在最新的repo编译，似乎不能完成了。
`Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ssr-plus:
 *      v2ray-plugin
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
package/Makefile:65: recipe for target 'package/install' failed
`
看起来像是和这个commit有关系：
`commit d96a2e78f7095f9d11c9abbfae487b4c8ef8b7d9
Author: coolsnowwolf <coolsnowwolf@gmail.com>
Date:   Sun Apr 18 09:55:04 2021 +0800

    v2ray-plugin: move source
`
我使用的是coolsnowwolf的repo，来编译。
ideas?

Thanks,

kenzok8, 2021-04-19:
是你的问题吧，我云编译能正常呢

jiahanbryant, 2021-04-19:
hmmm...我前两天（4月17日）编译都还正常，我今天运行了：
·./scripts/feeds update -a && ./scripts/feeds install -a·
然后再`make defconfig`就会产生这个提示：
```
WARNING: Makefile 'package/feeds/kenzo/luci-app-passwall/Makefile' has a dependency on 'v2ray-plugin', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'v2ray-plugin', which does not exist
```
编译就会失败。或者我再试试。。

jiahanbryant, 2021-04-19:
@kenzok8 你可以重现吗？因为我的还是有这个问题。

chenyg0911, 2021-04-22:
comments the line  which need V2ray-plugin in:
`$Lededir/feeds/helloworld/luci-app-ssr-plus/Makefile` & `$Lededir/feeds/kenzo/luci-app-passwall/Makefile` 
[image]

then Run: ./scripts/feeds update -a && ./scripts/feeds install -a

jiahanbryant, 2021-04-24:
> comments the line which need V2ray-plugin in:
> `$Lededir/feeds/helloworld/luci-app-ssr-plus/Makefile` & `$Lededir/feeds/kenzo/luci-app-passwall/Makefile`
> [image]
> 
> then Run: ./scripts/feeds update -a && ./scripts/feeds install -a

我没有仔细check这个build的architecture,如果我把这个comment out，意味着我没有来v2ray feature？

chenyg0911, 2021-04-25:
lean默认用的xray客户端，应该没影响。我用vless正常。

jiahanbryant, 2021-04-25:
> lean默认用的xray客户端，应该没影响。我用ssrp vless正常。passwall没试过。

Thanks bro, just to be clear, so you still have the option when trying to create a new server node?
[image]

chenyg0911, 2021-04-25:
> > lean默认用的xray客户端，应该没影响。我用ssrp vless正常。passwall没试过。
> 
> Thanks bro, just to be clear, so you still have the option when trying to create a new server node?
> [image]

[image]

jiahanbryant, 2021-04-25:
> > > lean默认用的xray客户端，应该没影响。我用ssrp vless正常。passwall没试过。
> > 
> > 
> > Thanks bro, just to be clear, so you still have the option when trying to create a new server node?
> > [image]
> 
> [image]

ok 谢谢。这个早晚要修复的吧，我关闭了。

closed 2021-04-25 by jiahanbryant