kenzok8/openwrt-packages #148: luci.mk file  错误

doumeki, 2021-04-21:
RROR: please fix package/feeds/kenzo/luci-app-aliddns/Makefile - see logs/package/feeds/kenzo/luci-app-aliddns/dump.txt for details
ERROR: please fix package/feeds/kenzo/luci-app-gost/Makefile - see logs/package/feeds/kenzo/luci-app-gost/dump.txt for details

--------------------------------------------------

Makefile:22: ../../luci.mk: No such file or directory
make[1]: *** No rule to make target '../../luci.mk'.  Stop.

kenzok8, 2021-04-21:
rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig
如果再出错，我就回退

doumeki, 2021-04-22:
是目录错误吧。看其它的app是这样写的
$(TOPDIR)/feeds/luci/luci.mk

insomniacdoll, 2021-04-24:
报错+1
实测修改成

$(TOPDIR)/feeds/luci/luci.mk
或者
../../luci/luci.mk

即可

closed 2021-04-26 by doumeki