kenzok8/openwrt-packages #150: 编译最新的naiveproxy出错

king1393, 2021-04-26:
Applying ./patches/001-Remove-concurrency-limit.patch using plaintext: 
patching file src/net/tools/naive/naive_proxy.cc
Hunk #2 FAILED at 38.
1 out of 2 hunks FAILED -- saving rejects to file src/net/tools/naive/naive_proxy.cc.rej
patching file src/net/tools/naive/naive_proxy_bin.cc
patch unexpectedly ends in middle of line
Patch failed!  Please fix ./patches/001-Remove-concurrency-limit.patch!
Makefile:141: recipe for target '/lede/build_dir/target-x86_64_musl/naiveproxy-90.0.4430.85-4/.prepared_06d10f6c68d683b3ab3f7c94e9bea458_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/lede/build_dir/target-x86_64_musl/naiveproxy-90.0.4430.85-4/.prepared_06d10f6c68d683b3ab3f7c94e9bea458_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/lede/feeds/kenzo/naiveproxy'
time: package/feeds/kenzo/naiveproxy/compile#1.49#1.03#2.26
package/Makefile:111: recipe for target 'package/feeds/kenzo/naiveproxy/compile' failed
make[2]: *** [package/feeds/kenzo/naiveproxy/compile] Error 2
make[2]: Leaving directory '/lede'
package/Makefile:107: recipe for target '/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/lede'
/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

kenzok8, 2021-04-26:
fix typo error

king1393, 2021-04-26:
OK，已完成编译

closed 2021-04-26 by king1393