kenzok8/openwrt-packages #155: 前几天更新后出现了adguardhome的报错！

superabbite1, 2021-05-17:
`Collecting package info: done
WARNING: Makefile 'package/openwrt-packages/adguardhome/Makefile' has a build dependency on 'node-yarn/host', which does not exist
WARNING: Makefile 'package/openwrt-packages/adguardhome/Makefile' has a build dependency on 'packr/host', which does not exist
Installing all packages from feed packages.`

[image]

doumeki, 2021-05-18:
me tooo
include ../../lang/golang/golang-package.mk 这条也不对。
还有为什么要改adguardhome的大小写？

my5xhni, 2021-05-23:
我也遇到这个问题了，你们怎么解决的？

the55555day, 2021-05-24:
> include ../../lang/golang/golang-package.mk

这条可以通过修改为绝对路径解决

dcdebug, 2022-02-11:
my feeds.conf.default ：
```
src-git packages https://github.com/coolsnowwolf/packages
src-git luci https://github.com/coolsnowwolf/luci
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git
#src-git video https://github.com/openwrt/video.git
#src-git targets https://github.com/openwrt/targets.git
#src-git oldpackages http://git.openwrt.org/packages.git
#src-link custom /usr/src/openwrt/custom-feed
```
when I ```./script/feeds install -a```的时候，出现:
```
....
WARNING: Makefile 'package/openwrt-packages/adguardhome/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/openwrt-packages/adguardhome/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/openwrt-packages/adguardhome/Makefile' has a build dependency on 'node-yarn/host', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'antfs-mount', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'wsdd2', which does not exist
WARNING: Makefile 'package/lean/baidupcs-web/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/small/brook/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/openwrt-packages/cdnspeedtest/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget-ssl', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'wget-ssl', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'wget-ssl', which does not exist
WARNING: Makefile 'package/lean/dnsproxy/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/openwrt-packages/filebrowser/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/openwrt-packages/filebrowser/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/lean/frp/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libgstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1-libs', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/go-aliyundrive-webdav/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/openwrt-packages/gost/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/small/hysteria/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/small/ipt2socks/Makefile' has a dependency on 'libuv', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/small/kcptun/Makefile' has a build dependency on 'golang/host', which does not exist
[... cut ...]
WARNING: Makefile 'package/small/v2ray-core/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/small/v2ray-plugin/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist
WARNING: Makefile 'package/small/xray-core/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/small/xray-plugin/Makefile' has a build dependency on 'golang/host', which does not exist
....
```
?  ? ?  what I should to do ?

closed 2023-02-01 by kenzok8