kenzok8/openwrt-packages #157: adguardhome依旧报错

Netflixxp, 2021-05-19:
package/Makefile:111: recipe for target 'package/feeds/kenzok/adguardhome/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed

jiawm, 2021-05-19:
一样报错，无法编译成功

superabbite1, 2021-05-19:
一直没有解决..............

tefiszx, 2021-05-19:
我也遇到了adguard编译报错。看提示是go语言支持找不到。
```
Makefile:41: ../../lang/golang/golang-package.mk: No such file or directory
gmake[1]: *** No rule to make target '../../lang/golang/golang-package.mk'.  Stop.

```

HK-Journalist, 2021-05-19:
把Makefile的41行改成`include $(TOPDIR)/feeds/packages/lang/golang/golang-package.mk`

jiawm, 2021-05-19:
Makefile:41: ../../lang/golang/golang-package.mk: No such file or directory
Makefile:94: WARNING: skipping adguardhome -- package has no install section
make[3]: *** No rule to make target '../../lang/golang/golang-package.mk'.  Stop.
make[3]: Leaving directory '/workdir/openwrt/package/kenzo/adguardhome'
make[2]: *** [package/kenzo/adguardhome/compile] Error 2
time: package/kenzo/adguardhome/compile#0.11#0.03#0.14
package/Makefile:111: recipe for target 'package/kenzo/adguardhome/compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

tefiszx, 2021-05-20:
> 
> 
> 把Makefile的41行改成`include $(TOPDIR)/feeds/packages/lang/golang/golang-package.mk`

推个PR上去啊

superabbite1, 2021-05-20:
我现在只能用他更新前的那个版本

Netflixxp, 2021-05-25:
已经解决了，使用下面命令后再重头编译一下
rm -rf ./feeds
make distclean

closed 2021-05-25 by Netflixxp