kenzok8/openwrt-packages #159: 昨天和今天编译openwrtclash都无法安装

sf2322571, 2021-05-20:
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-openclash:
 *      curl
 * opkg_install_cmd: Cannot install package luci-app-openclash.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/cc/openwrt19'
package/Makefile:110: recipe for target '/home/cc/openwrt19/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/cc/openwrt19/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/cc/openwrt19'
/home/cc/openwrt19/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

sf2322571, 2021-05-20:
去掉openwrtclash可以编译成功

lauglam, 2023-07-02:
我存在同样的问题
[image]

closed 2026-02-24 by kenzok8