kenzok8/openwrt-packages #170: 编译出错

la8896, 2021-06-17:
ERROR: package/feeds/kenzo/lua-maxminddb failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [world] Error 1
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
[image]

marshalshaodev, 2021-06-22:
我也遇到了，应该是有些包和op不兼容，特别是和lean的。

wggglggg, 2021-07-15:
> 我也遇到了，应该是有些包和op不兼容，特别是和lean的。

这种情况要如何处理?

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.