kenzok8/openwrt-packages #172: lua-maxminddb云编译报错

allok1985, 2021-06-18:
git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 93da9f4 Fixed implicit declaration of gai_strerror() compiler warning, that lead to occational segfualts, i.e. if the address we attempt to look up is invalid
Packing checkout...
touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.prepared_ce0c972e45f42506dd9eb8cf9cab3f68_6664517399ebbbc92a37c5bb081b5c53_check
. /workdir/openwrt/include/shell.sh; gzip -dc /workdir/openwrt/dl/lua-maxminddb-0.2.tar.gz | tar -C /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2
touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.prepared_ce0c972e45f42506dd9eb8cf9cab3f68_6664517399ebbbc92a37c5bb081b5c53
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.configured_*
rm -f /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.lua-maxminddb_installed
touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built_check
MAKEFLAGS="" /workdir/openwrt/staging_dir/host/bin/ninja  -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2 
ninja: Entering directory `/workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2'
ninja: error: loading 'build.ninja': No such file or directory
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built] Error 1
Makefile:54: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built' failed
make[3]: Leaving directory '/workdir/openwrt/feeds/kenzok8/lua-maxminddb'
time: package/feeds/kenzok8/lua-maxminddb/compile#0.30#0.08#1.68
    ERROR: package/feeds/kenzok8/lua-maxminddb failed to build.
make[2]: *** [package/feeds/kenzok8/lua-maxminddb/compile] Error 1
package/Makefile:114: recipe for target 'package/feeds/kenzok8/lua-maxminddb/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
package/Makefile:110: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make: *** [world] Error 2
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
Error: Process completed with exit code 2.

Lionzj, 2021-06-21:
同样的问题

facerleo, 2021-06-23:
我的也是这样

la8896, 2021-06-24:
替換lua-maxminddb便可
https://github.com/jerrykuku/lua-maxminddb

Lionzj, 2021-06-24:
> 替換lua-maxminddb便可
> https://github.com/jerrykuku/lua-maxminddb

正解! 我的自动编译现在就是用的这个
https://github.com/Lionzj/openwrt/blob/main/diy-part1.sh

dd-bpir3, 2024-06-18:
> 替換lua-maxminddb便可 https://github.com/jerrykuku/lua-maxminddb

怎么替换

closed 2024-06-20 by kenzok8