kenzok8/openwrt-packages #177: passwall缺少依赖

lxwq, 2021-07-05:
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'brook', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'chinadns-ng', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'trojan-go', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'trojan-plus', which does not exist
WARNING: Makefile 'package/feeds/helloworld/naiveproxy/Makefile' has a build dependency on 'ninja/host', which does not exist

lxwq, 2021-07-06:
重新拉了一下所有代码,缺少的依赖更多了,
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'ipt2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'brook', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'chinadns-ng', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'dns2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'pdnsd-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'shadowsocks-rust-sslocal', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'shadowsocksr-libev-ssr-local', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'shadowsocksr-libev-ssr-redir', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'shadowsocksr-libev-ssr-server', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'simple-obfs', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'trojan-go', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'trojan-plus', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-passwall/Makefile' has a dependency on 'v2ray-plugin', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'dns2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'pdnsd-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-check', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocks-rust-sslocal', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocks-rust-ssserver', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-local', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-redir', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-server', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'simple-obfs', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'trojan', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'ipt2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-ssr-plus/Makefile' has a dependency on 'v2ray-plugin', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-vssr/Makefile' has a dependency on 'pdnsd-alt', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-vssr/Makefile' has a dependency on 'shadowsocksr-libev-ssr-local', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-vssr/Makefile' has a dependency on 'shadowsocksr-libev-ssr-redir', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-vssr/Makefile' has a dependency on 'shadowsocksr-libev-ssr-check', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-vssr/Makefile' has a dependency on 'simple-obfs', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-vssr/Makefile' has a dependency on 'trojan', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-vssr/Makefile' has a dependency on 'ipt2socks', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-vssr/Makefile' has a dependency on 'xray-plugin', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-vssr/Makefile' has a dependency on 'shadowsocksr-libev-ssr-server', which does not exist

Boos4721, 2021-07-07:
[https://github.com/R619AC-OpenWrt/OpenWrt-Packages](url)

kenzok8, 2021-07-08:
passwall依赖库下载链接，注意！在openwrt或者lean源码下编译passwall，要下载此依赖库
请自己看看清楚

closed 2021-07-09 by lxwq