kenzok8/openwrt-packages #191: 用LEDE的编译出错.用的虚拟机ubuntu20.04

showme2005, 2021-08-18:
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/alen/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/alen/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/shadowsocks/v2ray-plugin" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/shadowsocks/v2ray-plugin/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/alen/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.41.0" GO_BUILD_DIR="/home/alen/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.41.0/.go_work/build" GO_BUILD_BIN_DIR="/home/alen/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.41.0/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/alen/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/alen/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.41.0=v2ray-plugin-4.41.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/alen/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/alen/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/alen/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/alen/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.41.0=v2ray-plugin-4.41.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/alen/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/alen/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/alen/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.41.0/.go_work/build" GOCACHE="/home/alen/lede/tmp/go-build" GOMODCACHE="/home/alen/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/alen/lede/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1629214236' -linkmode external -extldflags '-L/home/alen/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/alen/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-s -w -X main.VERSION=v4.41.0 -buildid '1629214236' -linkmode external -extldflags '-L/home/alen/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/alen/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets
go: github.com/v2fly/v2ray-core/v4@v4.41.0: Get "https://proxy.golang.org/github.com/v2fly/v2ray-core/v4/@v/v4.41.0.mod": EOF

Building targets
no Go files in /home/alen/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.41.0/.go_work/build

make[3]: *** [Makefile:75: /home/alen/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.41.0/.built] Error 1
make[3]: Leaving directory '/home/alen/lede/feeds/small/v2ray-plugin'
time: package/feeds/small/v2ray-plugin/compile#0.26#0.15#5.02
    ERROR: package/feeds/small/v2ray-plugin failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/v2ray-plugin/compile] Error 1
make[2]: Leaving directory '/home/alen/lede'
make[1]: *** [package/Makefile:110: /home/alen/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/alen/lede'
make: *** [/home/alen/lede/include/toplevel.mk:230：world] 错误 2

以上的日志...

zhangguanzhang, 2021-09-17:
代理不强，再试，或者hack下添加goproxy再便宜

kenzok8, 2021-09-19:
make download -j8 V=s && find dl -size -1024c -exec ls -l {} \; && find dl -size -1024c -exec rm -f {} \;

make download && make -j$(nproc) V=s

第一个命令是拉取dl，对于0字节的等于是失败的会自动删除
第二个命令是防止没有拉取完成，再一次download，然后编译，这样如果是代理或者网络不稳定，没下载下来的能检查出来，除非你的扶墙是特别不ok

closed 2026-02-24 by kenzok8