kenzok8/openwrt-packages #198: adguardhome 出错

lukemin, 2021-09-08:
ERROR: please fix package/feeds/kenzo/adguardhome/Makefile - see logs/package/feeds/kenzo/adguardhome/dump.txt for details
'''Package: adguardhome
Version: 0.106.3-1
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread +ca-bundle
Conflicts: 
Menu-Depends: 
Provides: 
Section: net
Category: Network
Repository: kenzo
Title: Network-wide ads and trackers blocking DNS server
Maintainer: Dobroslaw Kijowski <dobo90@gmail.com>
Source: adguardhome-0.106.3.tar.xz
License: GPL-3.0-only
LicenseFiles: LICENSE.txt
Type: ipkg
Description: Free and open source, powerful network-wide ads and trackers blocking DNS server.
https://github.com/AdguardTeam/AdGuardHome
Dobroslaw Kijowski <dobo90@gmail.com>
@@

Makefile:37: ../../lang/golang/golang-package.mk: No such file or directory
make[2]: *** No rule to make target '../../lang/golang/golang-package.mk'.  Stop.'''

kenzok8, 2021-09-08:
试试最新的，我本地跑没问题

Neurotoxin0, 2021-09-11:
确实有问题，我的工作流也是三天前开始暴毙的 :(

kenzok8, 2021-09-11:
https://github.com/kenzok8/openwrt-packages/releases

kenzok8, 2021-09-11:
我编译好好的，如果本地看下dl文件夹里是否是adguardhome-0.107.0-b.9.tar.xz ，如果不是建议用make distclean命令，再辛苦来一次

Neurotoxin0, 2021-09-12:
看了下workflow的报错：

/home/runner/work/OpenWrt/OpenWrt/OP/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/info/adguardhome.postinst-pkg: line 2: /etc/init.d/AdGuardHome: No such file or directory
postinst script ./usr/lib/opkg/info/adguardhome.postinst has failed with exit code 127
make[2]: *** [package/Makefile:73: package/install] Error 1
make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/OP'

Neurotoxin0, 2021-09-12:
具体工作流：https://github.com/Neurotoxin0/OpenWrt/runs/3577184305?check_suite_focus=true

kenzok8, 2021-09-12:
本地编译的.config文件上传更新下工作流看看

jiawm, 2021-09-12:
编译错误adguardhome

Neurotoxin0, 2021-09-13:
本地重新配置config文件后上传workflow以后就可以成功编译了，可能config是少了个选项的问题
[image]

closed 2021-09-15 by kenzok8