kenzok8/openwrt-packages #205: 求助大佬，云编译失败，/adguardhome插件出错，如何破解

ZBZB7050, 2021-09-16:
Configuring kmod-bnx2.
Configuring luci-i18n-firewall-zh-cn.
Configuring luci-i18n-docker-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Setting flags for package libc to hold.
Setting flags for package libgcc1 to hold.
Setting flags for package libatomic1 to hold.
Setting flags for package libstdcpp6 to hold.
Setting flags for package libpthread to hold.
Setting flags for package librt to hold.
cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/root-x86 /home/runner/work/lede/lede/build_dir/target-x86_64_musl/root.orig-x86
/home/runner/work/lede/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/info/adguardhome.postinst-pkg: line 2: /etc/init.d/AdGuardHome: No such file or directory
postinst script ./usr/lib/opkg/info/adguardhome.postinst has failed with exit code 127
make[2]: *** [package/Makefile:73: package/install] Error 1
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [package/Makefile:111: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

kenzok8, 2021-09-16:
make distclean重新来一次
dl文件夹里看下是否拉取了adguardhome-0.107.0-b.11.tar.xz

ZBZB7050, 2021-09-17:
大佬，我是用github云编译的，请问应该在openwrt-ci.yml里如何修改解决这个问题？（之前都是成功的）

zhangguanzhang, 2021-09-17:
再触发下试试

kenzok8, 2021-09-17:
.config 本地make menconfig再生成下，上传云编译目录

ZBZB7050, 2021-09-17:
好的，我试一下。十分感谢！

crazydayli, 2021-09-19:
我本地编译也遇到adguardhome失败的问题，看了一下dl文件夹里拉取的是adguardhome-0.106.3.tar.xz

kenzok8, 2021-09-19:
1、运行

smdx, 2024-01-07:
> 1、运行make distclean （ make clean仅仅是清除之前编译的可执行文件及配置文件。 而make distclean要清除所有生成的文件。） 2、git pull && ./scripts/feeds update -a && ./scripts/feeds install -a 3、再maken menuconfig选择好，重新跑一次 4、如果是云编译（第三步后运行./scripts/diffconfig.sh >diffconfig把生成的diffconfig改名后上传云编译的仓库地址下替换）

云编译第三步后把生成的diffconfig adguardhome内容替换到仓库编译还是出错

```
make[3]: Entering directory '/workdir/openwrt/feeds/kenzo/adguardhome'
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/workdir/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/workdir/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/AdguardTeam/AdGuardHome" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/AdguardTeam/AdGuardHome" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43" GO_BUILD_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43/.go_work/build" GO_BUILD_BIN_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43=adguardhome-0.107.43 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43=adguardhome-0.107.43 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43/.go_work/build" GOCACHE="/workdir/openwrt/tmp/go-build" GOMODCACHE="/workdir/openwrt/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash /workdir/openwrt/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1704450625' -linkmode external -extldflags '-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X github.com/AdguardTeam/AdGuardHome/internal/version.channel=release -X github.com/AdguardTeam/AdGuardHome/internal/version.version=v0.107.43 -X github.com/AdguardTeam/AdGuardHome/internal/version.buildtime=2024-01-05T10:30:25Z+0000 -X github.com/AdguardTeam/AdGuardHome/internal/version.goarm= -X github.com/AdguardTeam/AdGuardHome/internal/version.gomips= -buildid '1704450625' -linkmode external -extldflags '-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets
no required module provides package github.com/AdguardTeam/AdGuardHome: go.mod file not found in current directory or any parent directory; see 'go help modules'
Building targets
go: cannot find main module, but found .git/config in /workdir/openwrt
	to create a module there, run:
	cd ../../../../.. && go mod init

make[3]: *** [Makefile:84: /workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/kenzo/adguardhome'
time: package/feeds/kenzo/adguardhome/compile#0.11#0.04#0.14
    ERROR: package/feeds/kenzo/adguardhome failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/adguardhome/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
```

fairok, 2024-01-29:
> home-0.107.43/.built_check
> CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DI

一样的错误

closed 2024-03-23 by kenzok8