kenzok8/openwrt-packages #207: luci-app-adguardhome/root/etc/init.d/AdGuardHome错误

realconnor, 2021-09-20:
如题，日志报错，显示extra_command 命令不存在，自定义命令可以参考（[Init Scripts](https://openwrt.org/docs/techref/initscripts#custom_commands)），最新提交改动前的版本反而是没问题的。
```
-EXTRA_COMMANDS="do_redirect testbackup test_crontab force_reload isrunning"
-EXTRA_HELP="        do_redirect 0 or 1\
-        testbackup backup or restore\
-		test_crontab
-		force_reload
-		isrunning"

+extra_command "do_redirect" "0 or 1"
+extra_command "testbackup" "backup or restore"
+extra_command "test_crontab"
+extra_command "force_reload"
+extra_command "isrunning"
```

realconnor, 2021-09-20:
懒得发pr了，还有个问题，辛苦k兄改下
luci-app-adguardhome/luasrc/model/cbi/AdGuardHome/base.lua的41行
```
40		local tmp=luci.sys.exec(binpath.." --version | grep -m 1 -E 'v[0-9.]+' -o ")
41		version=string.sub(tmp, 1)
42		if version=="" then version="core error" end
```
需要改成
```
41		version=string.sub(tmp, 1, -2)
```
以去掉字符串末尾的换行符，以免出现下面这种奇怪的情况——version项换行了……
root@OpenWrt:~# cat /etc/config/AdGuardHome

config AdGuardHome 'AdGuardHome'
        option httpport '8080'
        option redirect 'none'
        option logfile '/tmp/AdGuardHome.log'
        option verbose '0'
        option hostsmd5 '5504bb04bd01cb0e340aee9e2ae065b2'
        option enabled '1'
        option waitonboot '0'
        option binpath '/usr/bin/AdGuardHome'
        option configpath '/etc/AdGuardHome/AdGuardHome.yaml'
        option workdir '/etc/AdGuardHome'
        option binmtime '1632036546'
        **option version 'v0.106.3
'**
        list old_redirect 'none'
        list old_port '6353'
        list old_enabled '1'

kenzok8, 2021-09-20:
刚刚修复了下，感谢你建议，就小修了一下，本地编译没啥问题
[image]

realconnor, 2021-09-20:
不好意思，你好像看错了…
不是adguardhome的问题，而是Luci界面luci-app-adguardhome，相关的启动脚本和lua网页脚本的问题。

jiawm, 2021-09-20:
[image]

realconnor, 2021-09-20:
> [image]
> 

你这应该是adh的问题（个人建议不勾选…直接在luci里下载官方binary），与luci界面无关，这俩相互之间没有依赖关系，我提的是后者的问题。

closed 2021-09-23 by realconnor