kenzok8/openwrt-packages #217: 应用商店 编译需要加什么依赖？luci-app-store

sxml, 2021-10-21:
WARNING: Makefile 'package/luci-app-store/Makefile' has a build dependency on 'app-store-ui/host', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.

whave, 2022-02-02:
无意中路过，楼上的兄弟，你差的这依赖与应用商店无关啊

SterlingCarter, 2023-08-03:
@kenzok8 麻烦大神更新一下主题包源码以适配istore文字显示居中和无图标问题嘛，超喜欢Netgear这款主题皮肤，貌似这个问题存在两年了，我们这种小白又不会，确实是无能为力啊，只能寄希望你这样的大神了

SterlingCarter, 2023-08-03:
@kenzok8 麻烦大神更新一下主题包源码以适配istore文字显示居中和无图标问题嘛，超喜欢Netgear这款主题皮肤，貌似这个问题存在两年了，我们这种小白又不会，确实是无能为力啊，只能寄希望你这样的大神了。

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.