kenzok8/openwrt-packages #221: gost无法自启

FengQianlong, 2021-11-04:
拉取lede最新的代码进行编译，编译成功
直接缝合您的插件库git clone https://github.com/kenzok8/openwrt-packages.git
除luci-app-gost外所选的插件都正常使用，但是就是gost无法自启动，手动可以启动成功，然后我看了/etc/init.d/gost 的权限也是755，我尝试把/etc/init.d/gost start 命令放入到后台启动项自定义启动中，但是一旦写入自定义启动后重启，连openwrt页面都无法打开，但是后台ssh可以连接，我打开gost文件看，如图所示
[image]
不知道是因为什么原因无法自启，还是说我编译过程中没有勾选某些选项，希望您能抽空给个解答，谢谢！

kenzok8, 2022-03-04:
检查权限
init.d文件夹下gost权限0775
因为我没安装大概路径是
chmod +x /etc/init.d/gost

azuo2021, 2022-03-06:
> 检查权限
> init.d文件夹下gost权限0775
> 因为我没安装大概路径是
> chmod +x /etc/init.d/gost

我确定了这个是有权限的。。

torny10, 2022-05-14:
我遇到了，解决了，脚本貌似有两个问题
1. start 时首先执行killall -9 "gost", 这会杀死脚本本身的进程  /etc/init.d/gost，Fix 将killall 参数 "gost" 改为 "/usr/bin/gost" 
2. 解决第一个问题后，我发现手动/etc/init.d/gost start 可以启动服务，但是系统启动执行时，也启动了进程，但启动之后进程就消失了，我是在脚本中增加sleep 10 从而观测到 gost进程启动时的样子的，后发现手动启动后，ssh终端中会不断冒出gost进程日志，感觉输出重定向有问题，我就改了 start中执行部分为：
     /usr/bin/gost ${run_command}  > /root/gost.log 2>&1 &
     echo "start $?"
然后就好了，第二点我不是特别理解为什么，可能时dropbear特殊之处？有人知道为啥请告知，感谢
除此之外 START改成了99 但这个应该没啥影响

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.

closed 2026-02-24 by kenzok8