kenzok8/openwrt-packages #227: passwall-naiveproxy failed to build.

haoranxu1989, 2021-11-19:
最近几日，passwall-naiveproxy不管是本地还是github云编译都报下面的错误。之前都是好的。

make[3]: *** [Makefile:130: /workdir/openwrt/dl/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/kenzo/naiveproxy'
time: package/feeds/kenzo/naiveproxy/compile#1.23#1.00#11.20
    ERROR: package/feeds/kenzo/naiveproxy failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/naiveproxy/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

tefiszx, 2021-11-20:
新版的naive需要一个chrome-linux的profdata文件，在默认的源码网站无法下载。所以失败了。本地编译能看到无法下载的文件名，可以从搜索引擎下载到本地后可以编译通过。

zhb7670, 2021-11-22:
用这个仓库，我也是这样子报错，可是我不想本地编译，大佬，云编译能解决吗？

zhb7670, 2021-11-22:
luci-app-passwall和 luci-app-ssr-plus下的插件naiveproxy 取消就可以编译成功！还请大佬解决这个问题，谢谢

tefiszx, 2021-11-23:
上游helloworld已经合并PR解决了，重新pull一下就可以了。[#Naiveissue722](https://github.com/fw876/helloworld/pull/722)

NorthBig, 2023-02-06:
是的，确实出现Naiveproxy failed to build.  还有  xray-plugin也是failed to build.  我试试不选择那个怎么样

closed 2023-02-11 by kenzok8