kenzok8/openwrt-packages #228: filebrowser编译不成功

rq1025330, 2021-11-22:
[image]
大佬这是什么原因呢

kenzok8, 2021-11-22:
是否选择了luci-app-advanced？高级设置里的filebrowser会编译冲突

rq1025330, 2021-11-22:
[image]
[image]
[image]
没有。

kenzok8, 2021-11-22:
[image]
我正常是不选择这个，你取消这个，试着编译下试试

rq1025330, 2021-11-22:
我这会取消试试，我以前没取消也可以用

rq1025330, 2021-11-23:
[image]
一样报错

kenzok8, 2021-11-23:
make distclean
git pull && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig
有时出错很玄学的，比如拉取dl文件时出错，怎么编译都过不了，我建议清除彻底，重新编译来一次

rq1025330, 2021-11-23:
hao ,我再试试

rq1025330, 2021-11-24:
[image]
大佬依然不行

rq1025330, 2021-11-24:
https://github.com/rq1025330/Actions-OP/blob/master/aarch64/.config
这是我本地编译的配置文件

rq1025330, 2021-11-24:
[image]
这个仓库的包和这个仓库
[image]
一样不，用那个好

rq1025330, 2021-11-24:
[image]

zzhzy21, 2021-12-31:
luci-app-filebrowser  应该还有一个18.06分支，lede固件需要使用18.06分支

closed 2023-01-30 by kenzok8