kenzok8/openwrt-packages #229: node编译错误-20211127

tujj99, 2021-11-27:
/home/tim/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/tim/openwrt/build_dir/target-x86_64_musl/node-v14.18.1/out/Release/obj.target/tools/v8_gypfiles/libv8_base_without_compiler.a: error adding symbols: malformed archive
collect2: error: ld returned 1 exit status
mkcodecache.target.mk:180: recipe for target '/home/tim/openwrt/build_dir/target-x86_64_musl/node-
[image]
v14.18.1/out/Release/mkcodecache' failed
make[5]: *** [/home/tim/openwrt/build_dir/target-x86_64_musl/node-v14.18.1/out/Release/mkcodecache] Error 1
rm 858f5e19acb9ecd22777e37548c1ab0c1f61f166.intermediate 92ff6655e2c3b17183ad2f3536dcc032bc536ae9.intermediate b0be9c8acb068026eadbdbffad802bb41569f071.intermediate 9f2a7dbff5e0dda2d3b2ab176814f143c31952e0.intermediate e125957625417897537871ec9c5b8e24a722c4e8.intermediate 68118ff57ca5681000af7e9e28d6ec6073fc5cb2.intermediate
Makefile:109: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/tim/openwrt/build_dir/target-x86_64_musl/node-v14.18.1'
Makefile:157: recipe for target '/home/tim/openwrt/build_dir/target-x86_64_musl/node-v14.18.1/.built' failed
make[3]: *** [/home/tim/openwrt/build_dir/target-x86_64_musl/node-v14.18.1/.built] Error 2
make[3]: Leaving directory '/home/tim/openwrt/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#15.60#0.79#16.49
    ERROR: package/feeds/packages/node failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 1
make[2]: Leaving directory '/home/tim/openwrt'
package/Makefile:110: recipe for target '/home/tim/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/tim/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tim/openwrt'
/home/tim/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

ken大，看一下这是啥问题？
[image]

MIAOBUSI, 2021-11-27:
同问。我也是一样的问题。

kenzok8, 2021-11-27:
我不知道你选择了什么插件，以前成功编译过固件吗？看下有没有安装g++

kenzok8, 2021-11-27:
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync

MIAOBUSI, 2021-11-27:
> sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7 msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync

之前是成功编译过的，g++安装了，node应该是网易云的插件需要的依赖

tujj99, 2021-12-22:
> 我不知道你选择了什么插件，以前成功编译过固件吗？看下有没有安装g++

排除法定位是jd-dailybonus的问题。
distclean后重新做一下配置就可以了。
-make distclean
git pull && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig

closed 2021-12-25 by tujj99

CitrusFlavored, 2022-01-05:
同,编译node版的网易云解锁就报这个错...