kenzok8/openwrt-packages #232: 能修复一下吗？adguardhome 一直有错。

dukmonki, 2021-12-05:
install -a 时，出错如下。

WARNING: Makefile 'package/feeds/kenzo/adguardhome/Makefile' has a build dependency on 'node-yarn/host', which does not exist
WARNING: Makefile 'package/feeds/kenzo/adguardhome/Makefile' has a build dependency on 'packr/host', which does not exist
Installing all packages from feed packages.

dukmonki, 2021-12-05:
全新安装的，一直有错。

dukmonki, 2021-12-11:
行么？

kenzok8, 2021-12-11:
明天我本地编译下，看下是否会出错，再解决

sbwml, 2021-12-13:
adg哪有这么麻烦，直接官方openwrt feed 下修改，每次build 永远是最新版本

```shell
# AdGuardHome Beta - Fix build with go17.x
pushd feeds/packages
adguardhome_version=`curl -s "https://api.github.com/repos/AdguardTeam/AdGuardHome/releases" | grep "tag_name" | head -n 1 | awk -F ":" '{print $2}' | sed 's/\"//g;s/,//g;s/ //g' | awk -F "v" '{print $2}'`
sed -ri "s/(PKG_VERSION:=)[^\"]*/\1$adguardhome_version/" net/adguardhome/Makefile
sed -i 's/release/beta/g' net/adguardhome/Makefile
sed -i 's/.*PKG_MIRROR_HASH.*/#&/' net/adguardhome/Makefile
sed -i '/init/d' net/adguardhome/Makefile
popd
```

sbwml, 2021-12-13:
> 明天我本地编译下，看下是否会出错，再解决

kenzok8/openwrt-packages repo里的 adguardhome 虽然是 b.15 tag，但同时 `$(AGH_VERSION_PKG).channel=release`  也应该被定义为  `$(AGH_VERSION_PKG).channel=beta` ，不然即使编译进固件也会无限被提示更新（0.16正式版），而0.17 beta 直接更新（退回）0.16 是会导致进程起不来的

wang1zhen, 2022-01-12:
本地编译可以复现同样的问题，建议更新一下

git-earth, 2022-10-28:
你这只是缺少依赖，自己补齐就行了

smdx, 2024-01-06:
> adg哪有这么麻烦，直接官方openwrt feed 下修改，每次build 永远是最新版本
>
> ```shell
> # AdGuardHome Beta - Fix build with go17.x
> pushd feeds/packages
> adguardhome_version=`curl -s [...]

大佬你好，云编译还是会出错

[1486](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1487)
make[3]: Entering directory '/workdir/openwrt/feeds/packages/net/adguardhome'
[1487](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1488)
mkdir -p /workdir/openwrt/dl
[1488](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1489)
SHELL= flock /workdir/openwrt/tmp/.adguardhome-0.107.43.tar.xz.flock -c '  	 /workdir/openwrt/scripts/dl_github_archive.py --dl-dir="/workdir/openwrt/dl" --url="https://github.com/AdguardTeam/AdGuardHome" --version="v0.107.43" --subdir="adguardhome-0.107.43" --source="adguardhome-0.107.43.tar.xz" --hash="x" || ( 	echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf adguardhome-0.107.43 && [ \! -d adguardhome-0.107.43 ] && git clone  https://github.com/AdguardTeam/AdGuardHome adguardhome-0.107.43 && (cd adguardhome-0.107.43 && git checkout v0.107.43 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd adguardhome-0.107.43 && git log -1 --format='\''@%ct'\''` && rm -rf adguardhome-0.107.43/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c adguardhome-0.107.43 | 	xz -zc -7e > /workdir/openwrt/tmp/dl/adguardhome-0.107.43.tar.xz && mv /workdir/openwrt/tmp/dl/adguardhome-0.107.43.tar.xz /workdir/openwrt/dl/ && rm -rf adguardhome-0.107.43; );     '
[1489](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1490)
adguardhome-0.107.43.tar.xz: Download from https://github.com/AdguardTeam/AdGuardHome failed
[1490](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1491)
adguardhome-0.107.43.tar.xz: Requires sha256sum for verification
[1491](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1492)
Checking out files from the git repository...
[1492](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1493)
Cloning into 'adguardhome-0.107.43'...
[1493](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1494)
Note: switching to 'v0.107.43'.
[1494](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1495)

[1495](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1496)
You are in 'detached HEAD' state. You can look around, make experimental
[1496](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1497)
changes and commit them, and you can discard any commits you make in this
[1497](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1498)
state without impacting any branches by switching back to a branch.
[1498](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1499)

[1499](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1500)
If you want to create a new branch to retain commits you create, you may
[1500](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1501)
do so (now or later) by using -c with the switch command. Example:
[1501](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1502)

[1502](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1503)
  git switch -c <new-branch-name>
[1503](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1504)

[1504](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1505)
Or undo this operation with:
[1505](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1506)

[1506](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1507)
  git switch -
[1507](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1508)

[1508](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1509)
Turn off this advice by setting config variable advice.detachedHead to false
[1509](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1510)

[1510](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1511)
HEAD is now at f6ad64bf all: fix chlog link
[1511](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1512)
Packing checkout...
[1512](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1513)
touch /workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43/.prepared_027ffd8c90bcd1f4a08e743b26335944_6664517399ebbbc92a37c5bb081b5c53_check
[1513](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1514)
. /workdir/openwrt/include/shell.sh; xzcat /workdir/openwrt/dl/adguardhome-0.107.43.tar.xz | tar -C /workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.43/.. -xf -
[... cut ...]
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
[1532](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1533)
make[1]: Leaving directory '/workdir/openwrt'
[1533](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1534)
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
[1534](https://github.com/smdx/X86_64-Actions/actions/runs/7430367114/job/20219975023#step:12:1535)
Error: Process completed with exit code 2.

closed 2026-02-24 by kenzok8