kenzok8/openwrt-packages #244: LEDE编译时 make -j8 download V=s 遇到'detached HEAD' 无法进行下一步！

KoWming, 2022-01-14:
emote: Total 12174 (delta 0), reused 0 (delta 0), pack-reused 12174
Receiving objects: 100% (12174/12174), 14.70 MiB | 1.14 MiB/s, done.
Resolving deltas: 100% (8196/8196), done.
Note: switching to 'd63ff863800a5645aca4309d5dd5962bd1e95543'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at d63ff86 Finish 2.5.3 ... fix some small build issue in last version
Packing checkout...
100 1010k    0 1010k    0     0  30394      0 --:--:--  0:00:34 --:--:--     0make[2]: Leaving directory '/home/weir/lede/feeds/small/shadowsocksr-libev'
time: package/feeds/small/shadowsocksr-libev/download#10.08#4.02#33.34

卡在这里 无法进行下一步，请教如何解决！感谢！

KoWming, 2022-01-14:
shadowsocksr-libev-2.5.6.tar.xz: Download from https://github.com/shadowsocksrr/shadowsocksr-libev failed
shadowsocksr-libev-2.5.6.tar.xz: Wrong hash (probably caused by .gitattributes), expecting 34308ed827a5dd4f4e35619914102d55b00604faa44fda051d1d25fb4a319325, got 1517b5a3c21521163fe8047c3dfc1f97ee0dfdb4a8377380119173b217784119
Checking out files from the git repository...
Cloning into 'shadowsocksr-libev-2.5.6'...
100 1010k    0 1010k    0     0  76929      0 --:--:--  0:00:13 --:--:--     0remote: Enumerating objects: 12174, done.
100 1010k    0 1010k    0     0  71588      0 --:--:--  0:00:14 --:--:--     0Receiving objects:100 1010k    0 1010k    0     0  66945      0 --:--:--  0:00:15 --:--:--     0Receiving objects:100 1010k    0 1010k    0     0  62867      0 --:--:--  0:00:16 --:--:--     0Receiving objects:100 1010k    0 1010k    0     0  59261      0 --:--:--  0:00:17 --:--:--     0Receiving objects:100 1010k    0 1010k    0     0  56044      0 --:--:--  0:00:18 --:--:--     0Receiving objects:remote: Total 12174 (delta 0), reused 0 (delta 0), pack-reused 12174
Receiving objects: 100% (12174/12174), 14.70 MiB | 2.92 MiB/s, done.
Resolving deltas: 100% (8196/8196), done.
Note: switching to 'd63ff863800a5645aca4309d5dd5962bd1e95543'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at d63ff86 Finish 2.5.3 ... fix some small build issue in last version
100 1010k    0 1010k    0     0  53157      0 --:--:--  0:00:19 --:--:--     0Packing checkout...
100 1010k    0 1010k    0     0  43086      0 --:--:--  0:00:24 --:--:--     0make[2]: Leaving directory '/home/weir/lede/feeds/small/shadowsocksr-libev'

kenzok8, 2022-01-14:
rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig

KoWming, 2022-01-14:
> rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig

大佬，还是无法继续下去，不知道是不是我科学网络的问题
[image]
。

kenzok8, 2022-01-15:
git reset --hard
git pull

sed -i '$a src-git small https://github.com/kenzok8/small' feeds.conf.default
sed -i '$a src-git kenzo https://github.com/kenzok8/openwrt-packages' feeds.conf.default

rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig

KoWming, 2022-01-16:
> git reset --hard git pull
>
> sed -i '$a src-git small https://github.com/kenzok8/small' feeds.conf.default sed -i '$a src-git kenzo https://github.com/kenzok8/openwrt-packages' feeds.conf.default
>
> rm [...]

感谢大佬，我在试试！

closed 2022-03-11 by kenzok8

KoWming, 2022-03-11:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。