kenzok8/openwrt-packages #253: luci-app-store插件缺少依赖

ligehhh, 2022-03-01:
WARNING: Makefile 'package/feeds/kenzok/luci-app-store/Makefile' has a dependency on 'luci-lib-ipkg', which does not exist

ligehhh, 2022-03-01:
不知道是作者的依赖还是lede的，大佬看看能不能修复

mumuou, 2022-03-02:
应该是这边的问题，最新代码也遇到了~~~

xztxy, 2022-03-02:
同样遇到问题了。。

WYC-2020, 2022-03-02:
改成luci-base

ligehhh, 2022-03-03:
> > 改成 luci-base
> 
> 下，在哪里改

package/feeds/kenzok/luci-app-store/Makefile这个配置文件里面把那个依赖改成上面说的那个

wow8szh, 2022-03-03:
> > > 改成 luci-base
> > 
> > 
> > 下，在哪里改
> 
> package/feeds/kenzok/luci-app-store/Makefile这个配置文件里面把那个依赖改成上面说的那个

打开了文件，还是不知道怎么改，你说详细一点吗？

ligehhh, 2022-03-03:
[image]
这样改

ligehhh, 2022-03-03:
sed -i 's/luci-lib-ipkg/luci-base/g' package/feeds/kenzok/luci-app-store/Makefile   这个命令也可以

xztxy, 2022-03-03:
> sed -i 's/luci-lib-ipkg/luci-base/g' package/feeds/kenzok/luci-app-store/Makefile 这个命令也可以

好的，感谢~！！

closed 2022-04-14 by kenzok8