kenzok8/openwrt-packages #255: adguardhome/files/adguardhome.init里的配置文件名字错误

starambler, 2022-03-03:
21 行
  procd_set_param command "$PROG" -c /etc/adguardhome.yaml -w "$WORK_DIR" --no-check-update

里面的/etc/adguardhome.yaml 应为AdGuardHome.yaml 。

kenzok8, 2022-03-03:
启动不了还是什么错误？

starambler, 2022-03-03:
luci里面 
默认配置文件   /etc/config/AdGuardHome.yaml
linux是区分大小写，所以将配置好的AdGuardHome.yaml文件上传的话，AdGuardHome程序不认。
若手动配置为adguardhome.yaml文件的话，当关掉AdGuardHome.程序再启动的话，又不认adguardhome.yaml

starambler, 2022-03-03:
实际上openwrt官方编译的默认配置文件也是adguardhome.yaml，但luci-app-addguardhome的默认配置文件为AdGuardHome.yaml。

cs-star, 2022-03-15:
之前只要把adguardhome启动禁用 把备份的配置文件复制到模版就可以使用 刚编译最新的 完了 AdGuardHome被默认禁用了 adguardhome不能重定向了

cs-star, 2022-03-15:
chmod 755 /etc/init.d/AdGuardHome && service AdGuardHome restart

cs-star, 2022-03-15:
addguardhome应该删除掉 不知道有啥用  每次编译完刷机后必须禁止启动 只留AdGuardHome

closed 2023-08-03 by kenzok8